alibaba/AliOS-Things #604: linkkitapp@esp8266下的aliosThings系统如何裁剪？

yunyinkeji, 2018-09-21:
目前编译下来发现默认的配置编译后内存占用太高了，修改哪里的配置，可以实现按需裁剪。

Making linkkitapp@esp8266.bin
Making linkkitapp@esp8266.hex
no definite address hint,using default mem configuration

                        AOS MEMORY MAP                            
|=================================================================|
| MODULE                                   | ROM       | RAM      |
|=================================================================|
| newlib_stub                              | 0         | 450      |
| alicrypto                                | 0         | 51655    |
| kv                                       | 0         | 3218     |
| common                                   | 0         | 254      |
| libiot_sdk_impl                          | 0         | 3353     |
| recovery                                 | 0         | 1935     |
| kernel_init                              | 0         | 49       |
| rhino                                    | 0         | 19075    |
| xz-embedded                              | 0         | 6954     |
| uOTA                                     | 0         | 48706    |
| libpp                                    | 0         | 62417    |
| cjson                                    | 0         | 2359     |
| libiot_http                              | 0         | 5117     |
| libgcc                                   | 0         | 1242     |
| linkkitapp                               | 0         | 19992    |
| activation                               | 0         | 730      |
| libmain                                  | 0         | 30214    |
| chip_code                                | 0         | 566      |
| vfs                                      | 0         | 2315     |
| esp8266                                  | 0         | 12621    |
| board_esp8266                            | 0         | 1036     |
| digest_algorithm                         | 0         | 326      |
| libcirom                                 | 0         | 47163    |
| yloop                                    | 0         | 2754     |
| iotx-hal                                 | 0         | 9244     |
| libnet80211                              | 0         | 45787    |
| libiot_alcs                              | 0         | 40091    |
| libiot_log                               | 0         | 1547     |
| libiot_dm                                | 0         | 85607    |
| xtensa                                   | 0         | 6600     |
| libhal                                   | 0         | 37       |
| mbedtls                                  | 0         | 31812    |
| libiot_utils                             | 0         | 27579    |
| libcrypto                                | 0         | 14236    |
| nbpatch                                  | 0         | 10796    |
| libiot_system                            | 0         | 5169     |
| libiot_cm                                | 0         | 17413    |
| espos                                    | 0         | 1130     |
| osal                                     | 0         | 953      |
| netmgr                                   | 0         | 2483     |
| log                                      | 0         | 223      |
| libawss                                  | 0         | 38329    |
| hal                                      | 0         | 494      |
| libwpa                                   | 0         | 17870    |
| liblwip                                  | 0         | 45412    |
| libiot_mqtt                              | 0         | 21135    |
| libphy                                   | 0         | 48623    |
| *fill*                                   | 0         | 4704     |
|=================================================================|
| TOTAL (bytes)                            | 0         | 801775   |
|=================================================================|

yunyinkeji, 2018-09-21:
@xiaowenxia1 
基于最新的 alios Things rel 2.0.0版本
AOS ： rel_2.0.0
MCU : ESP8266 4MB Flash

walker2048, 2018-09-21:
这个统计数据应该是错的，你刷到板子上，能运行么？还是根本不能刷到板子上？

yunyinkeji, 2018-09-21:
刷到板子上是可以运行的，但是加网是有问题的，加网只有刚开始的6~7秒有效，后面再加网就无效了。

xiaowenxia1, 2018-09-21:
@yunyinkeji 你是说配网嘛？

yunyinkeji, 2018-09-21:
是的，按键启动配网后，有日志输出的，但是模块接收不到任何相关加网数据。和这个BUG一致：https://github.com/alibaba/AliOS-Things/issues/467

closed 2022-03-01 by skylarCai