alibaba/AliOS-Things #616: 钉钉的交流群进不了，可以新建几个群么

BigBigWolf-AI, 2018-09-27:
**Describe the bug**

**environment**
1. branch(version): 
2. Desktop OS(windows linux or macOS):
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

librae8226, 2018-09-27:
可以，但现在还没建起来，对开源项目来说IM群多的缺点是不利于知识沉淀。
有问题先提到这里吧。

Dean0315, 2018-10-08:
> **Describe the bug**
> 
> **environment**
> 
> 1. branch(version):
> 2. Desktop OS(windows linux or macOS):
> 3. example:
> 
> **Expected behavior**
> 
> **Screenshots**
> 
> **Additional context**

AliOS Things技术交流群-Banana：23130352（钉钉搜索群号），新群已经有了

BigBigWolf-AI, 2018-10-08:
这个新群根本搜不到呀，用钉钉，不行你们试试

BigBigWolf-AI, 2018-10-08:
> > **Describe the bug**
> > **environment**
> > 
> > 1. branch(version):
> > 2. Desktop OS(windows linux or macOS):
> > 3. example:
> > 
> > **Expected behavior**
> > **Screenshots**
> > **Additional context**
> 
> AliOS Things技术交流群-Banana：23130352（钉钉搜索群号），新群已经有了

这个群用钉钉搜不到呀，搜不到呀，搜不到呀！！！

Dean0315, 2018-10-09:
> > > **Describe the bug**
> > > **environment**
> > > 
> > > 1. branch(version):
> > > 2. Desktop OS(windows linux or macOS):
> > > 3. example:
> > > 
> > > **Expected behavior**
> > > **Screenshots**
> > > **Additional context**
> > 
> > 
> > AliOS Things技术交流群-Banana：23130352（钉钉搜索群号），新群已经有了
> 
> 这个群用钉钉搜不到呀，搜不到呀，搜不到呀！！！

[image]

closed 2022-03-01 by skylarCai