alibaba/AliOS-Things #626: 怎么使用ESP8266的PWM功能？有没有相关的开发文档？

liefyuan, 2018-09-29:


realonlyme, 2018-09-30:
OS暂时没有对PWM功能做规范，目前如果使用的话请以芯片厂商提供为主，比如ESP8266：
你可以搜索下关键字"esp8266 PWM"能查到

xuhongv, 2018-10-08:
稍安勿躁，正在快马加鞭未下鞍测试demo，并发布！

liefyuan, 2018-10-31:
> 
> 
> 稍安勿躁，正在快马加鞭未下鞍测试demo，并发布！

这是我写的demo是基于SDK编程弄出来的，可以用，五路PWM正常输出。
`/*
 * Copyright (C) 2015-2017 Alibaba Group Holding Limited
 */

#include <aos/aos.h>
#include <hal/soc/soc.h>
#include "stdio.h"
#include <pwm.h>
#include <gpio.h>

/**
 * Brief:
 * This test code shows how to configure gpio and how to use gpio interrupt.
 *
 * GPIO status:
 * GPIO18: output
 * GPIO4: output
 * GPIO5:  input, pulled up, interrupt from rising edge and falling edge
 *
 * Test:
 * Connect GPIO18 with LED
 * Connect GPIO4 with GPIO5
 * Generate pulses on GPIO4, that triggers interrupt on GPIO5 to blink the led.
 *
 */

#define PWM_0_OUT_IO_MUX PERIPHS_IO_MUX_MTDI_U
#define PWM_0_OUT_IO_NUM 12
#define PWM_0_OUT_IO_FUNC FUNC_GPIO12

#define PWM_1_OUT_IO_MUX PERIPHS_IO_MUX_MTDO_U
#define PWM_1_OUT_IO_NUM 15
#define PWM_1_OUT_IO_FUNC FUNC_GPIO15

#define PWM_2_OUT_IO_MUX PERIPHS_IO_MUX_MTCK_U
#define PWM_2_OUT_IO_NUM 13
#define PWM_2_OUT_IO_FUNC FUNC_GPIO13

#define PWM_3_OUT_IO_MUX PERIPHS_IO_MUX_GPIO4_U
#define PWM_3_OUT_IO_NUM 4
#define PWM_3_OUT_IO_FUNC FUNC_GPIO4

#define PWM_4_OUT_IO_MUX PERIPHS_IO_MUX_MTMS_U
#define PWM_4_OUT_IO_NUM 14
#define PWM_4_OUT_IO_FUNC FUNC_GPIO14

#define PWM_CHANNEL 5

void ESP8266_PWM_RUN( void )
{
    /** PWM占空比变量 */
    static uint8_t set_duty = 0;

    /** 占空比加减标志 */
    static bool f = true;
    if ( f == true )
    {
        if ( ++set_duty >= 100 )
        {
            f = false;
        }
    }
    else
    {
        if ( --set_duty <= 0 )
        {
            f = true;
        }
    }

    /** 更新PWM通道0的占空比 */
    printf("set_duty>>>%d\r\n", set_duty);
    pwm_set_duty( set_duty, 0 );
    pwm_start();
}

// 输出一个固定的PWM波
void esp8266_pwm_output(uint32_t duty, uint8_t channel)
{
    pwm_set_duty(20, 0 );
    pwm_start();

    // pwm_set_duty(30, 1 );
    // pwm_start();

    pwm_set_duty(40, 2 );
    pwm_start();

    pwm_set_duty(50, 3 );
    pwm_start();

    pwm_set_duty(60, 4 );
    pwm_start();
}

/** 初始化PWM配置、系统定时器配置 */
void ESP8266_PWM_Init( void )
{
    struct pwm_param pwm_config_t;

    uint32_t io_info[ ][ 3 ]={
        {PWM_0_OUT_IO_MUX,PWM_0_OUT_IO_FUNC,PWM_0_OUT_IO_NUM},
        {PWM_1_OUT_IO_MUX,PWM_1_OUT_IO_FUNC,PWM_1_OUT_IO_NUM},
        {PWM_2_OUT_IO_MUX,PWM_2_OUT_IO_FUNC,PWM_2_OUT_IO_NUM},
        {PWM_3_OUT_IO_MUX,PWM_3_OUT_IO_FUNC,PWM_3_OUT_IO_NUM},
        {PWM_4_OUT_IO_MUX,PWM_4_OUT_IO_FUNC,PWM_4_OUT_IO_NUM},
    };

   pwm_config_t.duty[ 0 ] = 0;
   pwm_config_t.freq = 0;
   pwm_config_t.period = 100;

    pwm_init( pwm_config_t.period, pwm_config_t.duty, PWM_CHANNEL, io_info );

    // 输出一个固定的PWM波
    // pwm_set_duty(80, 0 );
    // pwm_start();
}

int application_start(int argc, char *argv[])
{
    ESP8266_PWM_Init();  // PWM 初始化

    esp8266_pwm_output(80, 0); // 输出固定PWM波

    aos_loop_run();

    return 0;
}
`

closed 2022-03-01 by skylarCai