alibaba/AliOS-Things #631: esp8266 build app-blink failed

saybabi, 2018-10-06:
**Describe the bug**
~~~
> Executing task: aos make blink@esp8266 <

aos-cube version: 0.2.60
Check if required tools for esp8266 exist
Build AOS Now
TOOLCHAIN_PATH=/home/xiaodi/work/AliOS-Things/build/compiler/gcc-xtensa-lx106/Linux64/bin/
Compiling auto_component
Making out/blink@esp8266/libraries/auto_component.a
Making blink@esp8266.elf
out/blink@esp8266/libraries/blink.a(blink.o):(.literal.app_trigger_low_action+0x8): undefined reference to `hal_gpio_output_low'
out/blink@esp8266/libraries/blink.a(blink.o):(.literal.app_trigger_high_action+0x4): undefined reference to `hal_gpio_output_high'
out/blink@esp8266/libraries/blink.a(blink.o):(.literal.gpio_isr_handler+0x14): undefined reference to `hal_gpio_input_get'
out/blink@esp8266/libraries/blink.a(blink.o):(.literal.gpio_isr_handler+0x18): undefined reference to `hal_gpio_output_toggle'
out/blink@esp8266/libraries/blink.a(blink.o):(.literal.application_start+0x4): undefined reference to `hal_gpio_init'
out/blink@esp8266/libraries/blink.a(blink.o):(.literal.application_start+0x8): undefined reference to `hal_gpio_enable_irq'
out/blink@esp8266/libraries/blink.a(blink.o): In function `app_trigger_low_action':
/home/xiaodi/work/AliOS-Things/app/example/blink/blink.c:46: undefined reference to `hal_gpio_output_low'
out/blink@esp8266/libraries/blink.a(blink.o): In function `app_trigger_high_action':
/home/xiaodi/work/AliOS-Things/app/example/blink/blink.c:52: undefined reference to `hal_gpio_output_high'
out/blink@esp8266/libraries/blink.a(blink.o): In function `gpio_isr_handler':
/home/xiaodi/work/AliOS-Things/app/example/blink/blink.c:39: undefined reference to `hal_gpio_input_get'
/home/xiaodi/work/AliOS-Things/app/example/blink/blink.c:40: undefined reference to `hal_gpio_output_toggle'
out/blink@esp8266/libraries/blink.a(blink.o): In function `application_start':
/home/xiaodi/work/AliOS-Things/app/example/blink/blink.c:63: undefined reference to `hal_gpio_init'
/home/xiaodi/work/AliOS-Things/app/example/blink/blink.c:70: undefined reference to `hal_gpio_init'
/home/xiaodi/work/AliOS-Things/app/example/blink/blink.c:77: undefined reference to `hal_gpio_init'
/home/xiaodi/work/AliOS-Things/app/example/blink/blink.c:80: undefined reference to `hal_gpio_enable_irq'
collect2: error: ld returned 1 exit status
build/aos_target_build.mk:337: recipe for target 'out/blink@esp8266/binary/blink@esp8266.elf' failed
make[1]: *** [out/blink@esp8266/binary/blink@esp8266.elf] Error 1
build/Makefile:265: recipe for target 'main_app' failed
make: *** [main_app] Error 2
[AliOS-Things] ERROR: "/" returned error code 2.
[AliOS-Things] ERROR: Command "/ h o m e / x i a o d i / w o r k / A l i O S - T h i n g s / b u i l d / c m d / l i n u x 6 4 / m a k e   H O S T _ O S = L i n u x 6 4   T O O L S _R O O T = / h o m e / x i a o d i / w o r k / A l i O S - T h i n g s / b u i l d   - e   - f   b u i l d / M a k e f i l e   b l i n k @ e s p 8 2 6 6" in "/home/xiaodi/work/AliOS-Things"
---
The terminal process terminated with exit code: 2
~~~
**environment**
1. branch(version): master
2. Desktop OS(windows linux or macOS):ubuntu16
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

librae8226, 2018-10-08:
这个是因为之前esp8266一直确实gpio的hal层对接。
现在可以参考 @xuhongv 提交的 #563 添加了对 esp8266 gpio 的支持（该PR已合入master）

saybabi, 2018-10-08:
good job! 
@librae8226 
@xuhongv

closed 2018-10-08 by saybabi

xuhongv, 2018-10-08:
@saybabi  esp8266的gpio口中断有点特殊，需要不仅仅在```hal```层，还要在业务层做点修改！