alibaba/AliOS-Things #641: 关于 SAL_SOCKET_OFFSET 0

virnop, 2018-10-10:
**Describe the bug**

文件sal_sockets.h第128行
`#define  SAL_SOCKET_OFFSET 0`

文件utils_net.c第44-47行
```
    pNetwork->handle = HAL_TCP_Establish(pNetwork->pHostAddress, pNetwork->port);
    if (0 == pNetwork->handle) {
        return -1;
    }
```

使用SAL第一次alloc_socket时得到的fd是0；导致utils_net.c认为联网失败。

**environment**
1. branch(version): rel_2.0.0
2. Desktop OS(windows linux or macOS): Windows 10 + IAR 8.3
3. example:  mqttapp

**Expected behavior**

**Screenshots**
[image]

**Additional context**

virnop, 2018-10-10:
@librae8226  你好，更改 SAL_SOCKET_OFFSET  的值又会导致
`#error SAL_SOCKET_OFFSET does not work with external FD_SET!`
麻烦帮忙看看我该从哪里下手修改。

lluwang, 2018-10-29:
@virnop 请问你的底层是如何对接的？这个地方的失败应该不是FD导致的。

virnop, 2018-10-30:
> @virnop 请问你的底层是如何对接的？这个地方的失败应该不是FD导致的。

@lluwang 感谢回答。底层是自己移植的一个SAL_Device，硬件是中移的M5310-A，至于SAL层，没有修改过。

lluwang, 2018-11-01:
请问SAL_Device能够成功创建TCP or UDP connection对吗？

virnop, 2018-11-15:
> 请问SAL_Device能够成功创建TCP or UDP connection对吗？

是的，可以创建。但是第一次创建socket返回的fd是0.

closed 2022-03-01 by skylarCai