alibaba/AliOS-Things #645: AliOS Studio 如何卸载干净? 为何我的 Atom 突然变成了AliOS Studio

cnahliu, 2018-10-14:
**Describe the bug**

       有一天下了个AliOS Studio ，后面也没怎么用。后面打开github 的Atom 编辑器，突然发现Atom 的界面变成了AliOS Studio，不是说 AliOS Studio 是基于 VSCode 开发的吗？不明白为什么会影响到Atom。
       
      后面自然是把 AliOS Studio 卸载掉，然后打开Atom发现还是AliOS Studio 的页面，然后又尝试把Atom 卸载掉重装，打开后竟然还是AliOS Studio的界面。真是感觉很疑惑。。。

**environment**
1. branch(version): 
2. Desktop OS(windows linux or macOS):macOS 10.14.1
3. example: 

**Expected behavior**

**Screenshots**
[image]

**Additional context**

xiaowenxia1, 2018-10-14:
@cnahliu 你的这个AliOS-Studio下载的可能是**AliOS**的对应的开发工具：http://www.alios.cn/aliosstudio ，并不是**AliOS-Things**的开发工具哈，AliOS-Things的开发工具(AliOS-Studio)在vscode 插件里安装:
[image]

cnahliu, 2018-10-14:
😔

closed 2018-10-14 by cnahliu