alibaba/AliOS-Things #657: esp8266,GPIO14 不停触发中断

taksing-leung, 2018-10-23:
**Describe the bug**
esp8266平台,不管初始化哪个GPIO.只要GPIO14接地就会不断触中断函数

**environment**
1. branch(version): 2.0
2. Desktop OS(windows linux or macOS):win10 docker
3. example: linkkitapp

**Expected behavior**

**Screenshots**

**Additional context**

SuperCoderMan, 2018-10-23:
因为这个工程里GPIO14是作为配网按键使用的。

SuperCoderMan, 2018-10-23:
> **Describe the bug**
> esp8266平台,不管初始化哪个GPIO.只要GPIO14接地就会不断触中断函数
> 
> **environment**
> 
> **Expected behavior**
> 
> **Screenshots**
> 
> **Additional context**

因为这个工程里GPIO14是作为配网按键使用的。

alexwjye, 2018-10-23:
试一下GPIO上电输入高电平

taksing-leung, 2018-10-24:
> > **Describe the bug**
> > esp8266平台,不管初始化哪个GPIO.只要GPIO14接地就会不断触中断函数
> > **environment**
> > **Expected behavior**
> > **Screenshots**
> > **Additional context**
> 
> 因为这个工程里GPIO14是作为配网按键使用的。

请问GPIO14是作为配网按键代码在哪个文件里?

taksing-leung, 2018-10-24:
> 试一下GPIO上电输入高电平
高低电平都会 不断触发中断

SuperCoderMan, 2018-10-24:
> > > **Describe the bug**
> > > esp8266平台,不管初始化哪个GPIO.只要GPIO14接地就会不断触中断函数
> > > **environment**
> > > **Expected behavior**
> > > **Screenshots**
> > > **Additional context**
> > 
> > 
> > 因为这个工程里GPIO14是作为配网按键使用的。
> 
> 请问GPIO14是作为配网按键代码在哪个文件里?

在AliOS\AliOS-Things\platform\mcu\esp8266\bsp\key.c中：
`#define KEY_GPIO_PIN GPIO_Pin_14
#define KEY_GPIO_MODE GPIO_Mode_Input
#define KEY_GPIO_PULLUP GPIO_PullUp_DIS
#define KEY_GPIO_INTRTYPE GPIO_PIN_INTR_NEGEDGE`

taksing-leung, 2018-10-25:
> 在AliOS\AliOS-Things\platform\mcu\esp8266\bsp\key.c中：
>
> ```
> #define KEY_GPIO_PIN GPIO_Pin_14
> #define KEY_GPIO_MODE GPIO_Mode_Input
> #define KEY_GPIO_PULLUP GPIO_PullUp_DIS
> #define KEY_GPIO_INTRTYPE [...]

代码里中断触发后没有清除一下中断,esp8266有点特殊

closed 2022-03-01 by skylarCai