alibaba/AliOS-Things #668: ubuntu使用vs-code烧写developerkit固件失败

wangym307, 2018-10-25:
```Making linkdevelop.device_ctrl@developerkit.hex
Making linkdevelop.device_ctrl@developerkit.bin

                        AOS MEMORY MAP
|=================================================================|
| MODULE                                   | ROM       | RAM      |
|=================================================================|
| newlib_stub                              | 250       | 0        |
| sal                                      | 11852     | 251      |
| alicrypto                                | 30938     | 8780     |
| kv                                       | 2883      | 36       |
| libc_nano                                | 17896     | 464      |
| vcall                                    | 3042      | 4        |
| kernel_init                              | 653       | 36       |
| yloop                                    | 1529      | 24       |
| rhino                                    | 12329     | 7629     |
| developerkit                             | 5086      | 1383     |
| framework                                | 256       | 12       |
| vfs                                      | 1803      | 1209     |
| digest_algorithm                         | 92        | 0        |
| libgcc                                   | 3120      | 0        |
| stm32l4xx_cube                           | 26342     | 8402     |
| device_ctrl                              | 2278      | 12       |
| cli                                      | 5414      | 481      |
| mqtt                                     | 12157     | 0        |
| netmgr                                   | 1659      | 245      |
| mbedtls                                  | 26350     | 76       |
| device_sal_bk7231                        | 8520      | 265      |
| sensor                                   | 9186      | 247      |
| iotkit                                   | 14584     | 1705     |
| log                                      | 406       | 20       |
| MQTTPacket                               | 1869      | 4        |
| hal                                      | 194       | 12       |
| atparser                                 | 5844      | 297      |
| armv7m                                   | 340       | 0        |
| *fill*                                   | 225       | 1606     |
|=================================================================|
| TOTAL (bytes)                            | 207097    | 33200    |
|=================================================================|
Build complete
Making .gdbinit
```
在ubuntu操作系统下安装了vscode以及alios-studio插件，编译环境已经搭建好，st-flash也可以正常调用，但是烧录时却提示找不到设备，使用`ls /dev/tty*`是可以看到有`/dev/ttyACM0`的，这说明pc应该已经连接上开发板了。
```
aos-cube version: 0.2.61
[INFO]: Target: linkdevelop.device_ctrl@developerkit
[INFO]: Currently in aos_sdk_path: '/home/wangym/ali/AliOS-Things'
scons: Reading SConscript files ...
[INFO]: Scons to upload!args:{'APPLICATION': 'linkdevelop.device_ctrl', 'COMMAND': 'upload', 'BOARD': 'developerkit'}

[INFO]:target: linkdevelop.device_ctrl@developerkit

[INFO]:Currently in aos_sdk_path: '/home/wangym/ali/AliOS-Things'

[INFO]:upload_cmd: '/home/wangym/ali/AliOS-Things/build/cmd/linux64/st-flash', host_os:'Linux64'

[INFO]:image_path: '/home/wangym/ali/AliOS-Things/out/linkdevelop.device_ctrl@developerkit/binary/linkdevelop.device_ctrl@developerkit.bin'

[INFO]:exec_cmd: '['/home/wangym/ali/AliOS-Things/build/cmd/linux64/st-flash', '--reset', 'write', '/home/wangym/ali/AliOS-Things/out/linkdevelop.device_ctrl@developerkit/binary/linkdevelop.device_ctrl@developerkit.bin', '0x08000000']'

st-flash 1.5.0
2018-10-25T16:09:01 WARN usb.c: Couldn't find any ST-Link/V2 devices
[ERROR]:firmware upload failed!

scons: done reading SConscript files.
scons: Building targets ...
scons: `.' is up to date.
scons: done building targets.```

librae8226, 2018-10-26:
插上板子打开命令行执行 `lsusb` 能看到什么？
另外确认下 ttyACM0 是不是它，插上板子和不插上板子，看看 /dev/tty* 设备变化？

wangym307, 2018-10-26:
> 插上板子打开命令行执行 `lsusb` 能看到什么？
> 另外确认下 ttyACM0 是不是它，插上板子和不插上板子，看看 /dev/tty* 设备变化？

通过插拔板子确认ttyACM0确实是developkit。   
后来发现烧录不进去是因为vscode没有超级用户权限，通过sudo启动vscode已经解决问题，谢谢！

closed 2018-10-26 by wangym307

librae8226, 2018-10-26:
其实把这个加上就好了
https://github.com/texane/stlink/tree/master/etc/udev/rules.d
不推荐 sudo，会引起一些不必要的问题