alibaba/AliOS-Things #674: @damon-yun 了解。我们正在完善这方面支持。

xiaofengZhangg, 2018-10-29:
目前可以参考到 `build/scripts/keil.py` 和 `build/scripts/iar.py`，分别用于自动从makefile生成keil和iar的工程。由于这个功能还处于开发阶段，所以虽能使用，但仍有一些待解决的问题。
简单的使用方式如：
```
aos make  helloworld@b_l475e IDE=iar
```
这样会在 `out/helloworld@b_l475e/iar_project` 中生成相应的 iar 工程。

以 iar 为例，具体实现机制是在 `aos_target_config.mk` 中生成 `build/scripts/config_mk.py`, 将编译相关的信息转化成 python 的数据结构，然后调用 `build/scripts/iar.py` 来生成对应的 iar工程文件 .ewp, .eww

生成工程的过程是全自动的，可以用 keil 或 iar 打开工程以后人工检查一遍。
感兴趣可以先体验一下。

_Originally posted by @librae8226 in https://github.com/alibaba/AliOS-Things/issues/541#issuecomment-418261132_

xiaofengZhangg, 2018-10-29:
按照你说的方法后出现如下问题请问怎么解决呢？
PS F:\AliOS-Things-master> aos make helloworld@frdmkl26z IDE=iar
aos-cube version: 0.2.60
Build AOS Now
TOOLCHAIN_PATH=
Compiling auto_component
process_begin: CreateProcess(NULL, iccarm --vla -e --dlib_config=full -D_TIMESPEC_DEFINED --silent --only_stdout --no_warnings --diag_warning=Pe167,Pe144,Pe513 -DSYSINFO_PRODUCT_MODEL=\"ALI_AOS_MKL26Z4\" -DSYSINFO_DEVICE_NAME=\"MKL26Z4\" -DSYSINFO_ARCH=\"Cortex-M0\" -DSYSINFO_MCU=\"mkl26z4\" -DCONFIG_NO_TCPIP -DFRDM_KL26Z -DFREEDOM -DTEST_CONFIG_KV_ENABLED=0 -DTEST_CONFIG_YLOOP_ENABLED=1 -DCPU_MKL26Z128VLH4 -Wall -fno-common -ffunction-sections -fdata-sections -ffreestanding -fno-builtin -mthumb -mapcs -std=gnu99 -mcpu=cortex-m0plus -mfloat-abi=soft -MMD -MP -Wno-format -Wno-incompatible-pointer-types -DSYSINFO_KERNEL_VERSION=\"AOS-R-2.0.0\" -DSYSINFO_APP_VERSION=\"app-1.0.0-20181029.1058\" -ggdb -Os -Wall -Wfatal-errors -fsigned-char -ffunction-sections -fdata-sections -fno-common -std=gnu11 -DAOS_SDK_VERSION_MAJOR=3 -DAOS_SDK_VERSION_MINOR=2 -DAOS_SDK_VERSION_REVISION=3 -Iout/helloworld@frdmkl26z/resources/ -DPLATFORM=\"frdmkl26z\" -I./app/example/helloworld/./ -I./board/frdmkl26z/. -I./board/frdmkl26z/../../platform/mcu/mkl26z4/ -I./board/frdmkl26z/../../platform/mcu/mkl26z4/CMSIS/Include -I./board/frdmkl26z/../../platform/mcu/mkl26z4/drivers -I././platform/mcu/mkl26z4/../../arch/arm/armv6m/gcc/m0plus/ -I././osal/./mico/include -I././osal/./include
-I./kernel/init/include -I./kernel/yloop/include -I./tools/cli/include -I././platform/arch/arm/armv6m/gcc/m0/ -I./kernel/rhino/core/include -I./kernel/rhino/uspace/include -I./kernel/rhino/debug/include -I./kernel/rhino/hal/soc -I./kernel/rhino/./ -I./kernel/rhino/fs/kv/include -I./kernel/rhino/vfs/include -I././middleware/common/include -I././middleware/alink/cloud/include -I./utility/log/include -I./kernel/hal/include -I./kernel/hal/../rhino -I./kernel/hal/../rhino/hal/soc -I.//network/include -I./app/example/helloworld -I./kernel/hal/include/hal -I./kernel/hal/include/hal/soc -DBUILD_BIN -DAOS_NO_WIFI -DOSAL_RHINO -DAOS_LOOP
-DHAVE_NOT_ADVANCED_FORMATE -DCONFIG_AOS_CLI -DAOS_KV -DAOS_VFS -DAOS_FRAMEWORK_COMMON -DCONFIG_CLOUD -DAOS_HAL -DMCU_FAMILY=\"mkl26z4\" -D__FILENAME__=\"testcase_register.c\" -o out/helloworld@frdmkl26z/Modules/out/helloworld@frdmkl26z/auto_component/testcase_register.o out/helloworld@frdmkl26z/auto_component/testcase_register.c, ...) failed.
make (e=2): 系统找不到指定的文件。
make.exe[1]: *** [build/aos_target_build.mk:270: out/helloworld@frdmkl26z/Modules/out/helloworld@frdmkl26z/auto_component/testcase_register.o] Error 2
make.exe[1]: *** Waiting for unfinished jobs....
process_begin: CreateProcess(NULL, iccarm --vla -e --dlib_config=full -D_TIMESPEC_DEFINED --silent --only_stdout --no_warnings --diag_warning=Pe167,Pe144,Pe513 -DSYSINFO_PRODUCT_MODEL=\"ALI_AOS_MKL26Z4\" -DSYSINFO_DEVICE_NAME=\"MKL26Z4\" -DSYSINFO_ARCH=\"Cortex-M0\" -DSYSINFO_MCU=\"mkl26z4\" -DCONFIG_NO_TCPIP -DFRDM_KL26Z -DFREEDOM -DTEST_CONFIG_KV_ENABLED=0 -DTEST_CONFIG_YLOOP_ENABLED=1 -DCPU_MKL26Z128VLH4 -Wall -fno-common -ffunction-sections -fdata-sections -ffreestanding -fno-builtin -mthumb -mapcs -std=gnu99 -mcpu=cortex-m0plus -mfloat-abi=soft -MMD -MP -Wno-format -Wno-incompatible-pointer-types -DSYSINFO_KERNEL_VERSION=\"AOS-R-2.0.0\" -DSYSINFO_APP_VERSION=\"app-1.0.0-20181029.1058\" -ggdb -Os -Wall -Wfatal-errors -fsigned-char -ffunction-sections -fdata-sections -fno-common -std=gnu11 -DAOS_SDK_VERSION_MAJOR=3 -DAOS_SDK_VERSION_MINOR=2 -DAOS_SDK_VERSION_REVISION=3 -Iout/helloworld@frdmkl26z/resources/ -DPLATFORM=\"frdmkl26z\" -I./app/example/helloworld/./ -I./board/frdmkl26z/. -I./board/frdmkl26z/../../platform/mcu/mkl26z4/ -I./board/frdmkl26z/../../platform/mcu/mkl26z4/CMSIS/Include -I./board/frdmkl26z/../../platform/mcu/mkl26z4/drivers -I././platform/mcu/mkl26z4/../../arch/arm/armv6m/gcc/m0plus/ -I././osal/./mico/include -I././osal/./include
-I./kernel/init/include -I./kernel/yloop/include -I./tools/cli/include -I././platform/arch/arm/armv6m/gcc/m0/ -I./kernel/rhino/core/include -I./kernel/rhino/uspace/include -I./kernel/rhino/debug/include -I./kernel/rhino/hal/soc -I./kernel/rhino/./ -I./kernel/rhino/fs/kv/include -I./kernel/rhino/vfs/include -I././middleware/common/include -I././middleware/alink/cloud/include -I./utility/log/include -I./kernel/hal/include -I./kernel/hal/../rhino -I./kernel/hal/../rhino/hal/soc -I.//network/include -I./app/example/helloworld -I./kernel/hal/include/hal -I./kernel/hal/include/hal/soc -DBUILD_BIN -DAOS_NO_WIFI -DOSAL_RHINO -DAOS_LOOP
-DHAVE_NOT_ADVANCED_FORMATE -DCONFIG_AOS_CLI -DAOS_KV -DAOS_VFS -DAOS_FRAMEWORK_COMMON -DCONFIG_CLOUD -DAOS_HAL -DMCU_FAMILY=\"mkl26z4\" -D__FILENAME__=\"component_init.c\" -o out/helloworld@frdmkl26z/Modules/out/helloworld@frdmkl26z/auto_component/component_init.o out/helloworld@frdmkl26z/auto_component/component_init.c, ...) failed.
make (e=2): 系统找不到指定的文件。
make.exe[1]: *** [build/aos_target_build.mk:270: out/helloworld@frdmkl26z/Modules/out/helloworld@frdmkl26z/auto_component/component_init.o] Error 2
make.exe: *** [build/Makefile:274: main_app] Error 2
[AliOS-Things] ERROR: "F" returned error code 2.
[AliOS-Things] ERROR: Command "F : / A l i O S - T h i n g s - m a s t e r / b u i l d \ c m d / w i n 3 2 / m a k e . e x e   H O S T _ O S = W i n 3 2   T O
O L S _ R O O T = F : / A l i O S - T h i n g s - m a s t e r / b u i l d   - e   - f   b u i l d / M a k e f i l e   h e l l o w o r l d @ f r d m k l 2 6 z
 I D E = i a r" in "F:\AliOS-Things-master"

librae8226, 2018-11-01:
@xiaofengZhangg 您想了解些什么呢？这个功能还在演进中。

closed 2022-03-01 by skylarCai