alibaba/AliOS-Things #681: AliOS Things可以提供ESP8266的PWM功能的库函数支持吗？

liefyuan, 2018-10-31:
ESP8266平台需要AliOS Things支持PWM功能！

在目录：kernel\rhino\hal\soc\pwm.h中有如下几个函数

- int32_t hal_pwm_init(pwm_dev_t *pwm);
- int32_t hal_pwm_start(pwm_dev_t *pwm);
- int32_t hal_pwm_stop(pwm_dev_t *pwm);
- int32_t hal_pwm_para_chg(pwm_dev_t *pwm, pwm_config_t para);
- int32_t hal_pwm_finalize(pwm_dev_t *pwm);

但是在**ESP8266**上面都还不支持，也没有相应的开发文档，目前分支master,rel2.0,rel1.3.3等都不支持

**希望可以支持ESP8266使用AliOS Things的PWM库函数进行配置PWM和输出PWM。**

Jianlong0430, 2018-10-31:
谢谢反馈 @liefyuan 。esp8266来自乐鑫，目前基于他们提供的SDK（pwm_init， pwm_start等，已合入github master）可实现对PWM的占空比、频率的控制。AliOS Things已提供相应的HAL层接口提供给芯片厂家，但目前部分芯片厂家倾向于使用自己的binary+API接口的方式提供，而不是源代码方式。

librae8226, 2018-11-01:
收到，谢谢 @liefyuan 
这部分 hal 层对接暂缺。我们会加入 Todo，如果急需也欢迎 pull request 把它支持起来。

closed 2022-03-01 by skylarCai