alibaba/AliOS-Things #686: 关于Developer Kit与Starter Kit的SW调试问题

hujiaodigua, 2018-11-01:
1. 请问在Developer Kit与Starter Kit两款板子中，使用ST-link/v2调试时，信号时是通过板载的F103转发到L4吗（F103作为调试辅助MCU吗？F103在调试中的功能是什么？）。

2.两款板子都能支持SWO调试吗？（直接链接L4的Trace-SWO引脚吗？）。

感谢回答。

walker2048, 2018-11-01:
那个就是个ST-Link，因为常见的ST-Link也是用STM32F103C8来做的。平常ST-Link能干啥，他们就能干啥。你仔细看厂家给的电路图，上面标注的也是stlink

librae8226, 2018-11-01:
> 信号时是通过板载的F103转发到L4吗（F103作为调试辅助MCU吗？F103在调试中的功能是什么？）。

F103 就是 ST-link。
您是指 SWD 调试吧？两款板子都支持的。
Developer Kit 上您需要对一下原理图，引脚外接进行调试是否可行。
另外，板子支持stlink调试，为何需要引出呢？

hujiaodigua, 2018-11-01:
意思是开发板，板载ST-LINK/V2吗，我指的不是SWD调试，我指的是SW追踪调试，ST-LINK/V2是F103C8T6；需要引出SWO是我有特殊的调试要求

walker2048, 2018-11-01:
是的，板载了，你试试嘛。可以单步调试追踪，可以查寄存器，内存。具体你看看厂家的原理图

hujiaodigua, 2018-11-01:
我还没有采购板子，我主要关注的是哪一款板子的SWO口是引出可以使用的

librae8226, 2018-11-01:
了解。可以看这里 https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-Developer-Kit-Hardware-Guide

hujiaodigua, 2018-11-01:
我在原理图上看到Developer Kit上L4的NRST、SWO、TMS和TCK都引出了，在照片上看到SWD接口的排针了，作为SWD接口单独调试L4使用。
而在Starter Kit上，我没有看到L4的SWO口被引出可以单独使用。

librae8226, 2018-11-01:
是的。
好奇 `特殊的调试要求` 是什么 😄

hujiaodigua, 2018-11-01:
感谢回答。 也没有多么特殊，就是处理器的指令追踪。

hujiaodigua, 2018-11-03:
很可惜，再反复的查阅资料与观察产品图片后，我发现Developer Kit上L4引出的SW调试接口中不含SWO口。
所以现在我想请问，还有什么办法把Developer Kit上L4的SWO口引出吗？
我们目前在选定标准化的IoT操作系统通过进行指令追踪实验以评估系统的可靠性。

hujiaodigua, 2018-11-03:
我说的SWO口指的是L4的TRACE_SWO/JTDO/PB3口

hujiaodigua, 2018-11-03:
我不使用STM32 ST-LINK Utility 的SWO viewer

librae8226, 2018-11-08:
> 还有什么办法把Developer Kit上L4的SWO口引出吗？

这样只能自己飞线了，L4的SWO如下图黄点处，请确认一下：
[image]

hujiaodigua, 2018-11-08:
感激不尽，这是我需要的答案

closed 2022-03-01 by skylarCai