alibaba/AliOS-Things #691: mqttapp项目做MQTT 无证书 TCP 链接 返回[err] TCP or TLS Connection failed

wuweiit, 2018-11-06:
**Describe the bug**
mqttapp项目做MQTT 无证书 TCP 链接 返回[err] TCP or TLS Connection failed

硬件是ESP8266EX 1MB flash

**environment**
1. branch(version): 
master
2. Desktop OS(windows linux or macOS):
macOS
3. example:  

** LOG **
[dbg] IOT_MQTT_Construct(3098): sizeof(iotx_mc_client_t) = 180!
[inf] iotx_mc_init(2183): MQTT init success!
[dbg] iotx_net_connect(256): TCP Connect...
[dbg] connect_tcp(44): network is  192.168.88.176 1883
D: 41 HAL_TCP_Establish() | establish tcp connection with server(host=192.168.88.176 port=1883)
Wifi connected.
D: 80 HAL_TCP_Establish() | success to establish tcp, fd=0
[dbg] connect_tcp(46): network return 0
[err] TCP or TLS Connection failed return -1
[inf] iotx_mc_disconnect(2628): mqtt disconnect!
[inf] iotx_mc_release(2687): mqtt release!
mqtt_client|203 :: MQTT construct failed
[inf] LITE_dump_malloc_free_stats(593): WITH_MEM_STATS = 0
linkkit_main|341 :: out of sample!

MQTT服务是自己装的，分析了源代码中是创建TCP出错，是只能使用阿里平台的MQTT服务器么？我打算搭建内网的物联系统的。

xiaowenxia1, 2018-11-06:
@wuweiit 
* 你跑的哪个example呢？
* 服务器的ip地址是在哪里更改的？
* MQTT服务你自己装的，支持TLS连接吗？

wuweiit, 2018-11-06:
example是 app/mqttapp，修改代码如下：

    mqtt_params.port = 1883; //pconn_info->port;
    mqtt_params.host = "192.168.88.176";//pconn_info->host_name;
    mqtt_params.client_id = "123"; // ;pconn_info->client_id;
    mqtt_params.username = "marker";//pconn_info->username;
    mqtt_params.password = "123";//pconn_info->password;
    mqtt_params.pub_key = NULL; // pconn_info->pub_key;

根据我打印日志的中走，自装的MQTT不支持TLS。

MQTT使用的Docker部署的：
version: '2'
services:
  mqtt:
    image: spants/mqtt
    restart: always
    ports:
      - 1883:1883
      - 9001:9001
    #volumes:
    #  - /DATA/opt/data/mqtt/config:/config

xiaowenxia1, 2018-11-06:
看这log显示，底层hal层代码正常连接上tcp了：
80 HAL_TCP_Establish() | success to establish tcp, fd=0
但是之后iotx_mc_connect里面又提示tcp连接失败：
[err] TCP or TLS Connection failed return -1
应该这个过程有一些逻辑性的错误。

wuweiit, 2018-11-07:
utils_net.c 文件中 44行HAL_TCP_Establish()返回的是0 ，判断 pNetwork->handle = 0返回-1 就导致[err] TCP or TLS Connection failed
```
 pNetwork->handle = HAL_TCP_Establish(pNetwork->pHostAddress, pNetwork->port);
    utils_debug("network return %d", pNetwork->handle);
    
    if (0 == pNetwork->handle) {
        return -1;
    }
```
我尝试将 `if (0 == pNetwork->handle) {` 改为` if (0 != pNetwork->handle) {`,可以正常之下发布消息的代码，但是连接是无效的就程序不断的reconnection。

我觉得TCP连接的代码应该有bug的。

xiaowenxia1, 2018-11-07:
谢谢您帮我们指出错误！
**正常之下发布消息的代码，但是连接是无效的就程序不断的reconnection。** 这个有具体的log吗？

xiaowenxia1, 2018-11-07:
好的，找到问题就好，这个问题我们会在下一版本修复。