alibaba/AliOS-Things #698: vscode插件创建项目失败

chenyulun, 2018-11-12:
**Describe the bug**
vscode插件创建项目失败
安装依赖好像安装的是constants
使用的时候使用的是constant

**environment**
1. branch(version): 
2. Desktop OS(windows linux or macOS): macOS
3. example: 

**Expected behavior**

**Screenshots**

**Additional context**

Create Project Fail.Traceback (most recent call last): File "/Users/chenyulun/.aos/python-venv/bin/aos", line 7, in <module> from aos.aos import main File "/Users/chenyulun/.aos/python-venv/lib/python3.6/site-packages/aos/aos.py", line 26, in <module> from .util import * File "/Users/chenyulun/.aos/python-venv/lib/python3.6/site-packages/aos/util.py", line 10, in <module> from constant import * ModuleNotFoundError: No module named 'constant'

xiaowenxia1, 2018-11-12:
@chenyulun aos-cube只支持python2.7哈，你的当前环境是python3，所以导致冲突了，请使用python2安装aos-cube，比如如下指令：

chenyulun, 2018-11-12:
不能够吧！不支持Python3，我用pip2重新装一遍就行了吗

xiaowenxia1, 2018-11-12:
@chenyulun 你装个python2哈，然后用pip2装aos-cube

chenyulun, 2018-11-12:
➜  ~ pip -V
pip 18.1 from /Users/chenyulun/anaconda3/lib/python3.6/site-packages/pip (python 3.6)
是不是要先切这个？

chenyulun, 2018-11-12:
/Users/chenyulun/.aos/python-venv/lib下面的已经是python3.6

xiaowenxia1, 2018-11-12:
不明白你啥意思？看起来有可能python虚拟环境也是用python3装的，最好是删除掉python-venv重新再试试。

chenyulun, 2018-11-12:
我使用
➜  ~ which pip
/Users/chenyulun/anaconda3/bin/pip
➜  ~ conda create -n python27 python=2.7
切换环境了

xiaowenxia1, 2018-11-12:
那你现在还有啥问题不？

chenyulun, 2018-11-12:
没有删除干净依赖，还是读取了py3.6的东西，我本来对py不熟悉，我自己摸索着吧，谢谢了

closed 2018-11-13 by chenyulun