alibaba/AliOS-Things #701: master的uart例程无法成功编译

warman2000, 2018-11-12:
PS C:\Users\wander\.aos\AliOS-Things> aos make uart@esp32devkitc
aos-cube version: 0.2.60
Check if required tools for esp32devkitc exist
信息: 用提供的模式无法找到文件。
Build AOS Now
TOOLCHAIN_PATH=C:/Users/wander/.aos/AliOS-Things/build/compiler/gcc-xtensa-esp32/Win32/bin/
信息: 用提供的模式无法找到文件。
信息: 用提供的模式无法找到文件。
Compiling uart
In file included from ././platform/mcu/esp32/bsp/include/driver/include/driver/uart.h:24:0,
                 from app/example/uart/uart.c:17:
././platform/mcu/esp32/bsp/include/soc/esp32/include/soc/uart_struct.h:366:3: error: conflicting type qualifiers for 'uart_dev_t'
 } uart_dev_t;
   ^
In file included from app/example/uart/uart.c:16:0:
./kernel/hal/../rhino/hal/soc/uart.h:71:3: note: previous declaration of 'uart_dev_t' was here
 } uart_dev_t;
   ^
In file included from app/example/uart/uart.c:17:0:
././platform/mcu/esp32/bsp/include/driver/include/driver/uart.h:104:3: error: conflicting types for 'uart_config_t'
 } uart_config_t;
   ^
In file included from app/example/uart/uart.c:16:0:
./kernel/hal/../rhino/hal/soc/uart.h:65:3: note: previous declaration of 'uart_config_t' was here
 } uart_config_t;
   ^
app/example/uart/uart.c: In function 'application_start':
app/example/uart/uart.c:31:9: error: 'uart_dev_t {aka volatile struct <anonymous>}' has no member named 'port'
     uart.port = 1 //for esp8266 , when the port set is 1 ,then the uart1 ande uart2 is the same baud_rate , but the uart1 is for log
         ^
app/example/uart/uart.c:32:5: error: expected ';' before 'uart'
     uart.config = uartConfig;
     ^
app/example/uart/uart.c:33:19: warning: passing argument 1 of 'hal_uart_init' from incompatible pointer type [-Wincompatible-pointer-types]
     hal_uart_init(&uart);
                   ^
In file included from app/example/uart/uart.c:16:0:
./kernel/hal/../rhino/hal/soc/uart.h:81:9: note: expected 'uart_dev_t * {aka struct <anonymous> *}' but argument is of type 'volatile uart_dev_t * {aka volatile struct <anonymous> *}'
 int32_t hal_uart_init(uart_dev_t *uart);
         ^
app/example/uart/uart.c:40:32: warning: passing argument 1 of 'hal_uart_recv_II' from incompatible pointer type [-Wincompatible-pointer-types]
         ret = hal_uart_recv_II(&uart, &receive_bytes, 15, &recv_size, HAL_WAIT_FOREVER);
                                ^
In file included from app/example/uart/uart.c:16:0:
./kernel/hal/../rhino/hal/soc/uart.h:121:9: note: expected 'uart_dev_t * {aka struct <anonymous> *}' but argument is of type 'volatile uart_dev_t * {aka volatile struct <anonymous> *}'
 int32_t hal_uart_recv_II(uart_dev_t *uart, void *data, uint32_t expect_size,
         ^
app/example/uart/uart.c:46:27: warning: passing argument 1 of 'hal_uart_send' from incompatible pointer type [-Wincompatible-pointer-types]
             hal_uart_send(&uart,receive_bytes,recv_size,1000);
                           ^
In file included from app/example/uart/uart.c:16:0:
./kernel/hal/../rhino/hal/soc/uart.h:94:9: note: expected 'uart_dev_t * {aka struct <anonymous> *}' but argument is of type 'volatile uart_dev_t * {aka volatile struct <anonymous> *}'
 int32_t hal_uart_send(uart_dev_t *uart, const void *data, uint32_t size, uint32_t timeout);
         ^
make.exe[1]: *** [build/aos_target_build.mk:270: out/uart@esp32devkitc/Modules/app/example/uart/uart.o] Error 1
make.exe[1]: *** Waiting for unfinished jobs....
make.exe: *** [build/Makefile:274: main_app] Error 2
[AliOS-Things] ERROR: "C" returned error code 2.
[AliOS-Things] ERROR: Command "C : / U s e r s / w a n d e r / . a o s / A l i O S - T h i n g s / b u i l d \ c m d / w i n 3 2 / m a k e . e x e   H O S T _ O S = W i n 3 2   T O O L S _ R O O T = C : / U s e r s / w a n d e r / . a o s / A l i O S - T h i n g s / b u i l d   - e   - f   b u i l d / M a k e f i l e   u a r t @ e s p 3 2 d e v k i t c" in "C:\Users\wander\.aos\AliOS-Things"

xiaowenxia1, 2018-11-13:
uart这个example目前不可用哈，不好意思！！

warman2000, 2018-11-13:
@xiaowenxia1 好的，谢谢，已经在ESP32搞定了

closed 2018-11-13 by warman2000