alibaba/AliOS-Things #709: 构建丢失aos/uData.h

chenyulun, 2018-11-13:
**Describe the bug**
aos-cube version: 0.2.63
Check if required tools for developerkit exist
Build AOS Now
TOOLCHAIN_PATH=/Users/chenyulun/AliOS-Things-project/AliOS-Things/build/compiler/gcc-arm-none-eabi/OSX/bin/
Compiling auto_component
Compiling vcall
device/sensor/drv/drv_modbus_sensors.c:9:10: fatal error: aos/uData.h: No such file or directory
 #include "aos/uData.h"
          ^~~~~~~~~~~~~
compilation terminated.
make[1]: *** [out/helloworld@developerkit/Modules/device/sensor/drv/drv_modbus_sensors.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [main_app] Error 2
[AliOS-Things] ERROR: "/Users/chenyulun/AliOS-Things-project/AliOS-Things/build/cmd/osx/make" returned error code 2.
[AliOS-Things] ERROR: Command "/Users/chenyulun/AliOS-Things-project/AliOS-Things/build/cmd/osx/make HOST_OS=OSX TOOLS_ROOT=/Users/chenyulun/AliOS-Things-project/AliOS-Things/build -e -f build/Makefile helloworld@developerkit" in "/Users/chenyulun/AliOS-Things-project/AliOS-Things"

**environment**
1. branch(version): developer
2. Desktop OS(windows linux or macOS): macOS
3. example: 
aos make helloworld@developerkit
**Expected behavior**

**Screenshots**

**Additional context**
构建显示找不到aos/uData.h

xiaowenxia1, 2018-11-13:
@chenyulun helloworld的example用不到udata的，你`aos make clean` 一下，然后再编译试试。

chenyulun, 2018-11-13:
@xiaowenxia1 
没有用啊
➜  AliOS-Things git:(master) ✗ aos make clean
aos-cube version: 0.2.63
Cleaning...
Done
➜  AliOS-Things git:(master) ✗ aos make helloworld@linuxhost
aos-cube version: 0.2.63
Making config file for first time
processing components: helloworld linuxhost platform/mcu/linux osal init auto_component
app_version:app-1.0.0-20181113.1911
kernel_version:AOS-R-2.0.0
all components: helloworld linuxhost platform/mcu/linux osal init auto_component yloop cli log arch.linux middleware.common middleware.alink.cloud rhino rhino.vfs hal kernel.rhino.vfs.device
app_version:app-1.0.0-20181113.1911
kernel_version:AOS-R-2.0.0
Build AOS Now
TOOLCHAIN_PATH=
Compiling helloworld
Compiling linuximpl
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
make[1]: *** [out/helloworld@linuxhost/Modules/./platform/mcu/linux/main/arg_options.o] Error 1error:
unknown warning option '-Wno-unused-but-set-variable'; did you meanmake[1]:  *** Waiting for unfinished jobs....'-Wno-unused-const-variable'?

      [-Werror,-Wunknown-warning-option]
make[1]: *** [out/helloworld@linuxhost/Modules/./platform/mcu/linux/main/hw.o] Error 1
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
make[1]: *** [out/helloworld@linuxhost/Modules/./platform/mcu/linux/main/main.o] Error 1
make: *** [main_app] Error 2
[AliOS-Things] ERROR: "/Users/chenyulun/AliOS-Things-project/AliOS-Things/build/cmd/osx/make" returned error code 2.
[AliOS-Things] ERROR: Command "/Users/chenyulun/AliOS-Things-project/AliOS-Things/build/cmd/osx/make HOST_OS=OSX TOOLS_ROOT=/Users/chenyulun/AliOS-Things-project/AliOS-Things/build -e -f build/Makefile helloworld@linuxhost" in "/Users/chenyulun/AliOS-Things-project/AliOS-Things"
---

closed 2018-11-13 by chenyulun