alibaba/AliOS-Things #710: macOS构建helloword@linuxhost报错

chenyulun, 2018-11-13:
**Describe the bug**
构建helloword@developerkit没问题；
但是构建helloword@linuxhost就报错，我的是macOS系统

**environment**
1. branch(version): master
2. Desktop OS(windows linux or macOS): macOS
3. example: 
```shell
aos make helloworld@linuxhost
```
**Expected behavior**

**Screenshots**

**Additional context**

```shell
Executing task: aos make helloworld@linuxhost <

aos-cube version: 0.2.63
Build AOS Now
TOOLCHAIN_PATH=
Making out/helloworld@linuxhost/libraries/helloworld.a
Compiling linuximpl
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
errorerror: : unknownunknown  warningwarning  optionoption  '-Wno-unused-but-set-variable';'-Wno-unused-but-set-variable';  diddid  youyou  meanmean  '-Wno-unused-const-variable'?'-Wno-unused-const-variable'?

            [-Werror,-Wunknown-warning-option][-Werror,-Wunknown-warning-option]

error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
make[1]: *** [out/helloworld@linuxhost/Modules/./platform/mcu/linux/main/main.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [out/helloworld@linuxhost/Modules/./platform/mcu/linux/main/arg_options.o] Error 1
make[1]: *** [out/helloworld@linuxhost/Modules/./platform/mcu/linux/main/hw.o] Error 1
ar: @out/helloworld@linuxhost/libraries/helloworld.ar_opts: No such file or directory
make[1]: *** [out/helloworld@linuxhost/libraries/helloworld.a] Error 1
make: *** [main_app] Error 2
[AliOS-Things] ERROR: "/Users/chenyulun/AliOS-Things-project/AliOS-Things/build/cmd/osx/make" returned error code 2.
[AliOS-Things] ERROR: Command "/Users/chenyulun/AliOS-Things-project/AliOS-Things/build/cmd/osx/make HOST_OS=OSX TOOLS_ROOT=/Users/chenyulun/AliOS-Things-project/AliOS-Things/build -e -f build/Makefile helloworld@linuxhost" in "/Users/chenyulun/AliOS-Things-project/AliOS-Things"
---
终端进程已终止，退出代码: 2

终端将被任务重用，按任意键关闭。
```

xiaowenxia1, 2018-11-13:
@chenyulun 跟刚才的问题一样，先aos make clean一下试试哈。

chenyulun, 2018-11-13:
@xiaowenxia1 没有用啊
```shell
➜  AliOS-Things git:(master) ✗ aos make clean
aos-cube version: 0.2.63
Cleaning...
Done
➜  AliOS-Things git:(master) ✗ aos make helloworld@linuxhost
aos-cube version: 0.2.63
Making config file for first time
processing components: helloworld linuxhost platform/mcu/linux osal init auto_component
app_version:app-1.0.0-20181113.1911
kernel_version:AOS-R-2.0.0
all components: helloworld linuxhost platform/mcu/linux osal init auto_component yloop cli log arch.linux middleware.common middleware.alink.cloud rhino rhino.vfs hal kernel.rhino.vfs.device
app_version:app-1.0.0-20181113.1911
kernel_version:AOS-R-2.0.0
Build AOS Now
TOOLCHAIN_PATH=
Compiling helloworld
Compiling linuximpl
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
make[1]: *** [out/helloworld@linuxhost/Modules/./platform/mcu/linux/main/arg_options.o] Error 1error:
unknown warning option '-Wno-unused-but-set-variable'; did you meanmake[1]:  *** Waiting for unfinished jobs....'-Wno-unused-const-variable'?

      [-Werror,-Wunknown-warning-option]
make[1]: *** [out/helloworld@linuxhost/Modules/./platform/mcu/linux/main/hw.o] Error 1
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Werror,-Wunknown-warning-option]
make[1]: *** [out/helloworld@linuxhost/Modules/./platform/mcu/linux/main/main.o] Error 1
make: *** [main_app] Error 2
[AliOS-Things] ERROR: "/Users/chenyulun/AliOS-Things-project/AliOS-Things/build/cmd/osx/make" returned error code 2.
[AliOS-Things] ERROR: Command "/Users/chenyulun/AliOS-Things-project/AliOS-Things/build/cmd/osx/make HOST_OS=OSX TOOLS_ROOT=/Users/chenyulun/AliOS-Things-project/AliOS-Things/build -e -f build/Makefile helloworld@linuxhost" in "/Users/chenyulun/AliOS-Things-project/AliOS-Things"
---
```

xiaowenxia1, 2018-11-13:
@chenyulun linuxhost主要是在linux上编译，macOS没怎么验证过，就算能编译，编出来的linuxhost也应该执行不了哈。

chenyulun, 2018-11-13:
macOS是基于Unix的，cp,ls,ping,route等等命令都是想通的

xiaowenxia1, 2018-11-13:
@chenyulun 这些从unix上就有的基本命令是一样的，但是其他的地方很多不一样啊，动态库，gcc工具链等，上面的问题就是macOS上的工具链不支持`-Wno-unused-but-set-variable`的选项。

chenyulun, 2018-11-13:
嗯嗯，我不纠结了，我有远程centOS,有需要我去那边试试，谢谢你了