alibaba/AliOS-Things #713: master里的modbus_test例程无法编译

warman2000, 2018-11-16:
https://github.com/alibaba/AliOS-Things/tree/master/app/example/modbus_test的例程无法编译
信息如下：
PS C:\Users\wander\.aos\AliOS-Things> aos make modbus_test@esp32devkitc
aos-cube version: 0.2.60
Check if required tools for esp32devkitc exist
Making config file for first time
信息: 用提供的模式无法找到文件。
processing components: modbus_test esp32devkitc platform/mcu/esp32 osal init auto_component
kernel/rhino/bus/mbmaster/mbmaster.mk:21: *** no none-vfp gcc lib provided!.  Stop.
make.exe: *** No rule to make target 'out/modbus_test@esp32devkitc/config.mk', needed by 'main_app'.  Stop.
[AliOS-Things] ERROR: "C" returned error code 2.
[AliOS-Things] ERROR: Command "C : / U s e r s / w a n d e r / . a o s / A l i O S - T h i n g s / b u i l d \ c m d / w i n 3 2 / m a k e . e x e   H O S T _ O S = W i n 3 2   T O O L S _ R O O T =
C : / U s e r s / w a n d e r / . a o s / A l i O S - T h i n g s / b u i l d   - e   - f   b u i l d / M a k e f i l e   m o d b u s _ t e s t @ e s p 3 2 d e v k i t c" in "C:\Users\wander\.aos\AliOS-Things"
---

xiaowenxia1, 2018-11-16:
@warman2000 使用**mbmaster**需要支持浮点型哈，看`kernel/rhino/bus/mbmaster/mbmaster.mk`:

```makefile
ifeq ($(COMPILER),armcc)
$(error no armcc lib provided!)
else ifeq ($(COMPILER),iar)
$(error no iar lib provided!)
else ifeq ($(ENABLE_VFP),1)
$(NAME)_PREBUILT_LIBRARY := lib/$(HOST_ARCH)/gcc/mbmaster.a
else
$(error no none-vfp gcc lib provided!)
endif
```

xiaowenxia1, 2018-11-16:
目前mbmaster不支持esp32哈。

xiaowenxia, 2018-11-20:
@warman2000 udata主要是在developerkit上做的验证，但是也支持esp32，不过需要去做esp32上的传感器适配，具体请参考：[基于AliOS Things Developer Kit开发板的外接传感器开发教程](https://github.com/alibaba/AliOS-Things/wiki/%E5%9F%BA%E4%BA%8EAliOS-Things-Developer-Kit%E5%BC%80%E5%8F%91%E6%9D%BF%E7%9A%84%E5%A4%96%E6%8E%A5%E4%BC%A0%E6%84%9F%E5%99%A8%E5%BC%80%E5%8F%91%E6%95%99%E7%A8%8B)

closed 2022-03-01 by skylarCai