alibaba/AliOS-Things #716: 给studio的一些建议

hanfengcan, 2018-11-21:
一些改进建议:

* activationEvents改成onCommand激活，这样才不会每次都激活插件

```
	"activationEvents": [
		"onCommand:alios-studio.createProject"
	],
```

* createProject的defaultPath可以考虑使用`workspaceFolders`。如果打开一个文件夹则提示使用当前文件夹；

* 考虑支持sdk自定义路径。2G的东西放C盘，估计很多人会难受；

* 另外，考虑存在py2和py3的系统；

暂时就这些，工具比之前完善很多。不过，从安装到使用还不够开箱即用 😄

xiaowenxia, 2018-11-21:
这些建议都非常有价值，目前只在AliOS-Things目录/工程目录下才**激活插件**(或者只显示创建工程图标)的功能已经在计划中，同时`aliosStudio.sdkPath`也会做相应调整。**defaultPath**和**支持sdk自定义路径**这2个是非常好的改进哈，谢谢你的建议。

hanfengcan, 2018-11-21:
我看有 `isAosSource`这个配置项。 如果设置true的时候，创建工程不执行clone操作就可以了吧

hanfengcan, 2018-11-21:
另外， 多个python版本的话， 可以尝试 

`virtualenv.exe -p D:\Python27\python.exe` 通过-p来指定版本。

不过我尝试了，修改插件代码。执行的时候提示路径不存在。 可以能是`execCommandToOutput`的问题

hanfengcan, 2018-11-21:
还有，不用在插件启动的时候，update sdkpath。考虑改在创建的时候，一起创建配置文件

xiaowenxia, 2018-11-21:
> 我看有 `isAosSource`这个配置项。 如果设置true的时候，创建工程不执行clone操作就可以了吧

`isAosSource`跟创建工程没有直接关系哈，AliOS-Things的源码里面的`.vscode/settings.json`有`"aliosStudio.isAosSource": true`，代表当前是AliOS-Things的源码目录。

xiaowenxia, 2018-11-21:
> 另外， 多个python版本的话， 可以尝试
> 
> `virtualenv.exe -p D:\Python27\python.exe` 通过-p来指定版本。
> 
> 不过我尝试了，修改插件代码。执行的时候提示路径不存在。 可以能是`execCommandToOutput`的问题
> 
> 我在cmder和cmd中执行可以正常。 vscode就有问题

这个问题我去评估一下看看是否可行。

hanfengcan, 2018-11-21:
> > 我看有 `isAosSource`这个配置项。 如果设置true的时候，创建工程不执行clone操作就可以了吧
> 
> `isAosSource`跟创建工程没有直接关系哈，AliOS-Things的源码里面的`.vscode/settings.json`有`"aliosStudio.isAosSource": true`，代表当前是AliOS-Things的源码目录。

我发现有个.aos文件，里面有个AOS_SDK_PATH。改写一下路径就变成自定义啦 🤣

xiaowenxia, 2018-11-21:
是的，不过我们一般是直接打开AliOS-Things的目录进行开发。

hanfengcan, 2018-11-21:
> 是的，不过我们一般是直接打开AliOS-Things的目录进行开发。

问题是太卡了

walker2048, 2018-11-22:
在什么系统下？python2和python3可以考虑用conda管理环境，在vscode里实现不同目录使用不同的python环境，不难

walker2048, 2018-11-22:
不会用vitrualenv，不会搞python，之前大佬推荐conda，就学着用了。感觉还不错

hanfengcan, 2018-11-24:
写了篇文章介绍如何使用studio， https://www.jianshu.com/p/fbefd012429c

xiaowenxia, 2018-12-04:
非常感谢 :smile:

closed 2018-12-28 by xiaowenxia

lemualFan, 2019-07-05:
对于使用vscode开发，建议提示安装virualenv，在vscode里面一键安装aos-cube，完成环境配置。全局安装的aos-cube，vscode检测不到。