alibaba/AliOS-Things #724: windows版本的studio, update命令似乎有bug

hanfengcan, 2018-11-24:
**Describe the bug**

Windows的esptool是esptool.exe，不是esptool.py

**environment**
1. branch(version): 
2. Desktop OS(windows linux or macOS):Windows

**Screenshots**

```
Executing task: aos upload helloworld@esp32devkitc <

aos-cube version: 0.2.63
[INFO]: Target: helloworld@esp32devkitc
[INFO]: Currently in aos_sdk_path: 'F:\alios-things\AliOS-Things'

scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly
File "build\site_scons\site_init.py", line 39, in aos_global_config
scons: Reading SConscript files ...
[INFO]: Scons to upload!args:{'APPLICATION': 'helloworld', 'COMMAND': 'upload', 'BOARD': 'esp32devkitc'}

[INFO]: Target: helloworld@esp32devkitc
[INFO]: Currently in aos_sdk_path: 'F:\alios-things\AliOS-Things'

--- Available ports:
---  1: COM3                 u'Silicon Labs CP210x USB to UART Bridge (COM3)'
---  2: COM4                 u'com0com - serial port emulator (COM4)'
---  3: COM5                 u'com0com - serial port emulator (COM5)'
--- Enter port index or full name: 1
[INFO]: Running cmd:
        'esptool.py --chip esp32 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 F:\alios-things\AliOS-Things/platform/mcu/esp32/bsp/bootloader.bin 0x8000 F:\alios-things\AliOS-Things/platform/mcu/esp32/bsp/custom_partitions.bin 0x10000 F:\alios-things\AliOS-Things/out/helloworld@esp32devkitc/binary/helloworld@esp32devkitc.bin'
'esptool.py' 不是内部或外部命令，也不是可运行的程序
或批处理文件。
---host_os:Win32
[ERROR]: Firmware upload failed!

[AliOS-Things] ERROR: "scons" returned error code -1.
[AliOS-Things] ERROR: Command "scons -f F:\alios-things\AliOS-Things\build/ucube.py COMMAND=upload APPLICATION=helloworld BOARD=esp32devkitc" in "F:\alios-things\AliOS-Things"
---
```

**Additional context**

hanfengcan, 2018-11-26:
改成

`esptool --chip esp32 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 F:\alios-things\AliOS-Things/platform/mcu/esp32/bsp/bootloader.bin 0x8000 F:\alios-things\AliOS-Things/platform/mcu/esp32/bsp/custom_partitions.bin 0x10000 F:\alios-things\AliOS-Things/out/helloworld@esp32devkitc/binary/helloworld@esp32devkitc.bin`

xiaowenxia, 2018-11-26:
目前这个确实存在问题，谢谢你的建议。。

lwxyl, 2018-12-13:
> 改成
> 
> `esptool --chip esp32 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m ...`

在哪个文件里修改？

hanfengcan, 2018-12-14:
@lwxyl 直接在vs code的终端里面输入就行了.

xiaowenxia, 2018-12-14:
> > 改成
> > `esptool --chip esp32 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m ...`
> 
> 在哪个文件里修改？

如果要让`aos upload`指令或者AliOS Studio的upload按钮支持的话，需要更改build/site_scons/upload/esp32.json里面的文件哈。

closed 2022-03-01 by skylarCai