alibaba/AliOS-Things #735: windows 10  vscode build  慢

hgnid, 2018-11-29:
**Describe the bug**
点了build 大概一分钟 才开始编译

**environment**
1. branch(version): master
2. Desktop OS(windows linux or macOS):windows 
3. example: helloword

**Expected behavior**

**Screenshots**

**Additional context**

hgnid, 2018-11-29:
[image]

hgnid, 2018-11-29:
> 谢谢你给我们提的issues，能不能给一下当前windows的性能（ctrl+shift+p 然后 输入developer: startup performance可以查看详细的windows性能检测）

[image]

xiaowenxia, 2018-11-29:
收到，谢谢。

hgnid, 2018-11-29:
关闭了  Windows Defender 之后 问题解决，不知道为啥点编译，windows defender就狂扫描

xiaowenxia, 2018-11-29:
> 关闭了 Windows Defender 之后 问题解决，不知道为啥点编译，windows defender就狂扫描

关闭了 Windows Defender 之后，打开时间是多少秒呢？

hgnid, 2018-11-29:
> > 关闭了 Windows Defender 之后 问题解决，不知道为啥点编译，windows defender就狂扫描
> 
> 关闭了 Windows Defender 之后，打开时间是多少秒呢？

大概2秒后就开始编译了，不关defender 把 rg.exe 在defender设置中 配置成例外，好像也可以。

closed 2018-12-04 by xiaowenxia