alibaba/AliOS-Things #736: 在xr871上跑例程 simple_mqqt，Link Develop下的device_ctrl以及linkvoiceapp 均编译失败

mmabluesky, 2018-11-29:
**Describe the bug**
aos make simple_mqqt@xr871evb xr871=1 编译失败

**environment**
1. branch(version): dev
2. Desktop OS(windows linux or macOS):linux
3. example: 
simple_mqqt

**Expected behavior**
make: *** No rule to make target 'out/simple_mqqt@xr871evb/config.mk', needed by 'main_app'.  Stop.

**Screenshots**

[image]

**Additional context**

xiaowenxia, 2018-11-29:
> @xiaowenxia
> 我是切到developer分支 编译的 ，例程在xr871上都编译失败

你`simple_mqqt`敲错了，是`simple_mqtt`

mmabluesky, 2018-11-29:
@xiaowenxia 
不行 ，还是hal_spi_send_recv这个函数错误

xiaowenxia, 2018-11-29:
请使用master分支哈。

mmabluesky, 2018-11-29:
@xiaowenxia 
master分支上 ，我就一个helloworld编译成功了 ，
我想在master分支下，把simple_mqtt这个例子拷贝进去，
然后aos make simple_mqtt@xr871evb xr871=1，应该怎样做才能编译成功，
目前报这个
[image]

mmabluesky, 2018-11-29:
@xiaowenxia 
不行，改了没用的，mqttapp我都跑步起来，更不谈simple_mqtt了

[image]

xiaowenxia, 2018-11-30:
@mmabluesky xr871这款mcu比较特别，需要有image的规划配置文件，你可以参考一下新建一个simple_mqtt的cfg文件：`platform\mcu\xr871\image_cfg\`