alibaba/AliOS-Things #740: aos make mqttapp@xr871evb,烧录镜像后，串口连wifi一直连不上

mmabluesky, 2018-12-03:
**Describe the bug**
 编译mqttapp for xr871，烧录后，串口发命令netmgr connect ssid psswd， 显示一直连不上wifi，不停打印一些扫描信息，我能确定的是wifi名和密码都正确 ，

**environment**
1. branch(version):   master
2. Desktop OS(windows linux or macOS): linux
3. example: 

**Expected behavior**

**Screenshots**
[image]
[image]

**Additional context**

xiaowenxia, 2018-12-03:
@xinhuiding 关于这个问题，能不能帮忙看看哈，非常感谢。

mmabluesky, 2018-12-04:
@xiaowenxia 求助攻，大佬，编译，烧录都没问题就是不能联网，

mmabluesky, 2018-12-04:
@xinhuiding  最新的master分支上编译的，之前我仿例编写了device_ctrl这个例子的mk，然后编译了这个demo，想实现云端控制led的功能，发现配不了网，后面才想试试mqttapp的，发现同样的问题，连不上wifi

mmabluesky, 2018-12-04:
@xinhuiding 
你再试试撒，最新master分支，aos make mqttapp@xr871evb，

xiaowenxia, 2018-12-04:
@xinhuiding AliOS-Things的master现在更新比较频繁，确实可能存在问题。你们的AliOS-Things 的仓库：https://github.com/XradioTech/AliOS-Things  是否可以正常运行呢？

mmabluesky, 2018-12-04:
@xinhuiding  
你的https://github.com/XradioTech/AliOS-Things，mater分支下没问题，已经连上了，谢谢两位啦

xiaowenxia, 2018-12-04:
👍

mmabluesky, 2018-12-06:
@xinhuiding 
你们的mater分支下的linkkitapp一键配网测试过吗？  我编译烧录linkkitapp后，设备上电，打开阿里的OneAPk进行一键配网，怎么老是失败呢

mmabluesky, 2018-12-06:
@xiaowenxia @xinhuiding 
linkkitapp怎么一键配网老是失败，我用的是https://github.com/XradioTech/AliOS-Things中的master分支下的linkkitapp，编译烧录上电后，串口发命令 active_awss ，串口打，印显示active_awss  0或者1，然后启动one apk，进入配网V3_小智，死活搜索不到设备

mmabluesky, 2018-12-06:
@xiaowenxia 
大佬是我使用方式不对吗

xiaowenxia, 2018-12-06:
@mmabluesky 你这配网是用的啥手机app？

mmabluesky, 2018-12-06:
@xiaowenxia 
https：//open.aliplus.com/download/ 这个

mmabluesky, 2018-12-06:
在wiki页面看到的，

xiaowenxia, 2018-12-06:
不是这个app哈，在哪个wiki页面看到的？请使用这个页面里面的app：https://living.aliyun.com/doc#muti-app.html

xinhuiding, 2018-12-06:
是使用的XradioTech/AliOS-Things的ali_master分支吗， app使用https://living.aliyun.com/doc#muti-app.html  开发版或者国内用户版

mmabluesky, 2018-12-06:
@xinhuiding  
分支是XradioTech/AliOS-Things的master分支， app我是用那个智能家居app的国内版，还是一键配网失败了，我能确定的是linkkitapp里面的三元组我都正确填写了，app我扫描的也是智能生活开放平台我创建的产品的二维码，设备端productkey和我二维码里的productkey一致

xiaowenxia, 2018-12-06:
@mmabluesky 配网失败，具体是哪一步失败了呢？

mmabluesky, 2018-12-06:
@xiaowenxia 
app端填写wifi密码，搜索设备，设备端串口发命令启动嗅探分析，然后两端都没啥反应，联网失败

xiaowenxia, 2018-12-06:
@mmabluesky 先给设备端串口发命令： active_awss ，进入配网模式后，app端扫二维码，填wifi密码，然后就开始配网了，具体参考：https://living.aliyun.com/doc#wifi.html

mmabluesky, 2018-12-06:
@xiaowenxia 
是那样操作的，没用 ，

mmabluesky, 2018-12-07:
@xinhuiding 
大佬，你们最新mater分支，linkkitapp一键配网，可以吗？

xinhuiding, 2018-12-07:
使用的XradioTech/AliOS-Things的ali_master分支，不是master分支。或者dev_xr871_v1.3.3
分支

mmabluesky, 2018-12-07:
@xinhuiding 全部都搞定了，但是我还有个疑问，对于出产品，ali_master和dev_xr871_v1.3.3 ，哪个分支更为稳定呢