alibaba/AliOS-Things #742: 电源管理接口适配问题

meixuxunxue, 2018-12-05:
**Describe the bug**
在做电源管理接口适配时遇到个问题，不清楚是哪一步出的问题：
pwr_test 那个例子： count 值增长，但是 idle 的值一直保持非零值不变；
跟踪发现 board_cpu_pwr_init 执行，但是没有进过 board_cpu_c_state_set 函数。

**environment**
1. branch(version):  rel_2.0.0
2. Desktop OS(windows linux or macOS): windows
3. example: pwr_test

**Expected behavior**
idle 值增长

**Screenshots**
[image]

**Additional context**
无

liano1987, 2018-12-06:
idle值不涨可能是系统中有任务一直在运行，请确认一下系统是否能进idle任务。这个是适配在什么平台上的？

meixuxunxue, 2018-12-06:
rda5981x，pwr_test 中也没有别的任务啦。

meixuxunxue, 2018-12-06:
程序没有进  idle_task

liano1987, 2018-12-06:
不开启低功耗能进idle task吗？

meixuxunxue, 2018-12-06:
pwr_test 开启低功耗了的：
[image]
在wifi_init 里面加延时是会进入 idle_task 的，但是之后就进不去了，这块注掉之后，程序中就没有我们自己的任务了。
[image]

liano1987, 2018-12-06:
你先验证一下，把cpu_pwrmgmt_init();注释掉程序是否能正常进入idle_task。如果这个可以的话，那可能是适配的低功耗驱动有问题了

meixuxunxue, 2018-12-06:
不行，不能进入 idle_task 应该是哪里有任务在一直执行，但是我们自己主动创建的任务都注掉了。

liano1987, 2018-12-06:
不加低功耗也进不了idle_task的话，说明系统有任务一直在运行，或者有死循环。用CLI打印看看当前的任务

liano1987, 2018-12-06:
连接串口，在串口控制台输入tasklist会打印任务列表

meixuxunxue, 2018-12-06:
有两个额外的任务，看起来是正常的。
[image]

liano1987, 2018-12-07:
目前问题为系统中存在while循环导致系统无法进入idle

closed 2018-12-07 by liano1987