alibaba/AliOS-Things #743: 请问windows下安装aos出现这种问题该怎么办

yangyua, 2018-12-05:
[image]
[image]
已经安装了好几次，每次都出现这种问题，刚开始是因为安装了另外一个版本的python，卸载之后又按照提示的执行了python -m pip install --upgrade pip，但是打开VS Code之后也一直不会提示install aos-cube。
[image]

xiaowenxia, 2018-12-05:
你安装python的时候，有添加path吗：
[image]

yangyua, 2018-12-05:
安装Python的时候是添加了path的
[image]
这个是显示的pip是10.0.1

walker2048, 2018-12-06:
尝试运行下`pip list -vv`  
这个命令会显示包安装的具体路径，如果之前安装过其他版本的python，就会有部分包的路径是不一致的。
如下图，原先安装过python，后来卸载了
[image]
删除掉不同的目录后，再次运行`pip list -vv`，所有包的路径都正确了
[image]

walker2048, 2018-12-06:
如果你能正常使用编译，上传等功能，那就完全不需要在乎pip版本不是最新18.1这个提示，并没有什么影响

yangyua, 2018-12-06:
现在输入aos能识别命令了，包的路径应该也正确
[image]
但是VSCode编译时会有error
[image]

walker2048, 2018-12-06:
你装了git了没有，如果已安装，那就是自动下载工具链失败了，自己点击提示的链接下载，然后解压到错误提示里的路径（build/compiler/gccXXXXXX那个路径）

yangyua, 2018-12-06:
我把git卸载又重装了一遍之后编译成功了，十分感谢。