alibaba/AliOS-Things #760: 运行例子helloword,blink遇到问题

lurberic, 2018-12-14:
**Is your feature request related to a problem? Please describe.**
新人学习，按照教程，WIN10+PYTHON2.7.15+Git-2.20.0-64-bit，编译时出现如下
Compiling auto_component
Compiling kernel_init
In file included from ./include/aos/network.h:24:0,
                 from kernel/init/aos_init.c:10:
./device/sal/./include/sal_sockets.h:10:24: fatal error: sys/select.h: No such file or directory
 #include <sys/select.h>
**Describe the solution you'd like**
请问如何解决？多谢！

**Additional context**

xiaowenxia, 2018-12-14:
哪个分支？编译的啥指令呢？

lurberic, 2018-12-14:
点击AliOS studio build 
开始 是 git checkout master, 有人建议  git checkout developer ，还是不行，估计是配置问题，但是按照教程反复检查，并重复安装几次，还是不行

xiaowenxia, 2018-12-14:
编译的哪个APP ？哪个board呢？

lurberic, 2018-12-14:
编译例子中，Helloworld, blink ,板子是AliOSThingsDeveloperKit，昨天问技术群，改 git checkout master, 为git checkout Developer, 还是不行。。。。

lurberic, 2018-12-14:
已做了分支处理，操作界面的截图发到你邮箱了，请看看需要做什么初始化的配置，如方便远程操作一下，多谢！

lurberic, 2018-12-14:
[image]

xiaowenxia, 2018-12-14:
还是刚才说的，你有进行clean操作吗？helloworld是最基础的例程，一般不会存在编译问题。

lurberic, 2018-12-14:
clean 这些都做了好几次了

xiaowenxia, 2018-12-14:
> clean 这些都做了好几次了

当前你的代码分支可能有些问题，请输入`git status`查看一下当前代码是否有变动？如果有变动，如果有改动，需要输入`git checkout .`恢复所有改动。

lurberic, 2018-12-14:
按照你的提示，操作后，现象还是这样，估计系统与这些软件安装出现不兼容造成的，我在试试看吧

xiaowenxia, 2018-12-14:
编译问题应该和软件不兼容没有关系，也可以试试删除所有无关的文件：`git clean -df`，或者重新拉取一份代码。

closed 2022-03-01 by skylarCai