alibaba/AliOS-Things #766: ESP8266 怎么设置编译APP1 和APP2 做OTA用的设置？还有FLASH 类型选择？

Yvan623048750, 2018-12-17:


xiaowenxia, 2018-12-17:
@Yvan623048750 不太明白你具体啥需求？esp8266针对不同的flash会有不一样的flash分区，具体请参考：[board/esp8266/board.c](https://github.com/alibaba/AliOS-Things/blob/master/board/esp8266/board.c)。

Yvan623048750, 2018-12-18:
我的意思是，在alios-things工程编译里面怎么选择参数让它编译出不同FLASH型号的bin，app1  app2的BIN，现在默认编译出来的我不知道是什么类型的BIN，原来乐鑫官方ESP8266是选择FLASH参数APP参数才编译的，现在ALIOS-THINGS工程里没有找到选择参数编译的地方

wangym307, 2018-12-22:
你说的是app、qio/dout这些设置选项吧？在platform/mcu/esp8266/bsp/ESP8266_RTOS_SDK/Makefile里面，可以试着修改一下。

Lucifer3502, 2019-01-07:
@Yvan623048750 你的问题解决了吗？
@wangym307 不知道是alios-thing的问题，还是其它什么问题。直接编译V2.0.0的版本，在esp8266平台上运行直接崩溃无法运行。目标版选择的flash是1MB，makefile配置如下：
BOOT?=new
APP?=1
SPI_SPEED?=40
SPI_MODE?=QIO
SPI_SIZE_MAP?=2

崩溃的完整log：
[image]
[image]

simonliu009, 2019-02-07:
应该修改这个文件``.\build\site_scons\upload\esp8266.json``
文件内容如下：
```
{
    "cmd": [
        {
            "Linux32": "esptool.py", 
            "Linux64": "esptool.py", 
            "OSX": "esptool.py", 
            "Win32": "esptool.exe"
        }, 
        "--chip", 
        "esp8266", 
        "--port", 
        "@PORT@", 
        "--baud", 
        "921600", 
        "write_flash", 
        "--flash_size", 
        "detect", 
        "0x0", 
        "@AOSROOT@/platform/mcu/esp8266/bsp/boot_v1.7_921600.bin", 
        "0x3fc000", 
        "@AOSROOT@/platform/mcu/esp8266/bsp/esp_init_data_default.bin", 
        "0x3fe000", 
        "@AOSROOT@/platform/mcu/esp8266/bsp/blank.bin", 
        "0x1000", 
        "@AOSROOT@/out/@TARGET@/binary/@TARGET@.bin"
    ]
}
```

simonliu009, 2019-02-08:
根据此文资料：[Alios Things的Flash划分规则](https://yq.aliyun.com/articles/620022)
[image]
AliOS Things 由于功能需要，维护了一张flash分区表，这张表包括bootloader区、Application区、OTA TMP区以及parameters区。Application区和OTA TMP区交替运行固件，相当于原来的user1.bin和user2.bin交替运行。所以应该没有必要自己修改编译参数生成user1.bin和user2.bin了。

closed 2022-03-01 by skylarCai