alibaba/AliOS-Things #769: win10系统VS code使用alios-studio配置开发环境的bug

zongkai28, 2018-12-19:
**Describe the bug**
win10系统下，python已经预装，在命令行下可正常运行python和pip命令。
VS code安装alios-studio后提示："please install python and pip first"，非常令人费解。而且即便不理会这个提示，命令行手工安装aoc-cube等工具后，在后续用alios-studio编译时仍然会报python的异常。
检查发现alios-studio插件的extension.js文件中pip命令名竟然都是pip2，修改后就可以正常运行alios-studio了，而且压根都不需要再参考wiki上繁琐的环境配置流程，alios-studio全部帮忙搞定，包括后续安装aos-cube、编译器等等。
改动点如下：
[image]
alios-studio这么好，就不要因为这点小问题困扰初学者了。

**environment**
1. branch(version): alios.alios-studio-0.10.8
2. Desktop OS(windows linux or macOS):windows 10 x64
3. example: 

**Expected behavior**

**Screenshots**
[image]

**Additional context**

xiaowenxia, 2018-12-20:
@zongkai28  **非常感谢你的建议:smile:，目前使用pip2的原因是因为aos-cube只支持python2，如果pip指令是python3的pip的话，就会导致安装或者运行aos-cube错误。你的系统不支持pip2指令的话，应该是python2版本太旧了（`python** --version`），这个我们欠考虑，非常抱歉。后续我们也计划先识别当前python环境，然后再使用pip 安装。

> 根都不需要再参考wiki上繁琐的环境配置流程，alios-studio全部帮忙搞定，包括后续安装aos-cube、编译器等等。

谢谢你的赞同！:+1: AliOS-Studio的初衷就是最简化开发环境和流程，让开发者无门槛就能快速开始开发AliOS-Things。
如果你有关于AliOS-Studio更好的建议和想法的话，非常欢迎一起交流一下哈。:smile:

hanfengcan, 2018-12-20:
可以把 scripts 也加入系统路径

xiaowenxia, 2018-12-20:
> 可以把 scripts 也加入系统路径

@hanfengcan 你说的是啥scripts？

hanfengcan, 2018-12-21:
Python27\Scripts @xiaowenxia 

[image]

xiaowenxia, 2018-12-21:
@hanfengcan 是的，我们的[相关wiki文档](https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-Windows-Environment-Setup)里面也说明了要把python加入到Path环境变量里面的：
[image]

hanfengcan, 2018-12-21:
这个只是加 python 到系统路径, 不手动把 scripts加入系统路径的话, 会找不到pip

zongkai28, 2018-12-23:
为你们的热心回复赞一个👍
我确实是python初学者，一直偷懒用anaconda，所以系统里确实没找到过pip2.exe，这下又涨知识了👏

closed 2018-12-23 by zongkai28

xiaowenxia, 2018-12-29:
@hankchan 谢谢你的建议，关于 `alios studio增加对anaconda的支持` 我们会认真评估一下哈。

gwfighter, 2019-03-18:
第一次编译就出现错误，不知道是不是和这个有关，只是为了实现一个简单例程：helloworld
---------------------------------------------------------------------------------------------------------

> Executing task: aos make helloworld@aaboard_demo -c config <

aos-cube version: 0.3.1
Traceback (most recent call last):
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\aos\__main__.py", line 82, in main
    cli(None)
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\aos\commands\make.py", line 33, in cli
    make_build(make_args)
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\aos\commands\make.py", line 259, in make_build
    _run_make(['-e', '-f build/Makefile', make_args])
  File "c:\users\administrator\.aos\python-venv\lib\site-packages\aos\commands\make.py", line 285, in _run_make
    make_cmd_str = ' '.join([make_cmd, 'HOST_OS=' + host_os, 'TOOLS_ROOT=' + tools_dir] + list(arg_list))
  File "c:\users\administrator\.aos\python-venv\lib\encodings\utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xd6 in position 3: invalid continuation byte
[AliOS-Things] ERROR: Unknown Error: 'utf8' codec can't decode byte 0xd6 in position 3: invalid continuation byte