alibaba/AliOS-Things #775: win10的Ubuntu子系统下使用AliOS Things的问题

nostalgia-w, 2018-12-25:
1. 首先我在win10下安装了Ubuntu16.04的子系统
方法：设置——>更新和安全——>针对开发人员——>开发人员模式
控制面板——>程序——>程序和功能——>启用或关闭Windows功能——>适用于Linux的Windows子系统——>确定 （然后重启）
在应用商店中搜索Ubuntu，选择Ubuntu16.04，点击获取进行下载及安装

2. 在配置环境的时候，无法安装libss1-dev:i386
[image]

xiaowenxia, 2018-12-25:
@nostalgia-w WSL的ubuntu16.04版本和ubuntu16.04是有一点系统架构上的差异的，目前好像还没法安装libss1-dev:i386 这样的安装包。不过安装libss1-dev 这些包也是为了能够运行app@linuxhost 这样的代码，如果你不运行linuxhost的话，就不需要安装libss1-dev包，直接安装aos-cube就可以使用。

nostalgia-w, 2018-12-25:
好的，aos-cube安装没问题。
但还有一个问题。
我之前开发这个项目的时候是rel_1.3.1的版本，用的是另一台电脑的虚拟机，但是太卡了所以换成现在这台win10的电脑。所以我暂时用的AliOS Things的版本是rel_1.3.1的版本。

在使用命令 aos make helloworld@esp8266的时候，编译不成功。打印如下：

[image]
[image]

中间有一部分没截图

xiaowenxia, 2018-12-25:
WSL目前编译1.x.x的代码分支确实有这个问题，暂时还不知道什么导致的，但是master分支是正常的。

nostalgia-w, 2018-12-25:
嗯，用master分支编译的aos make helloworld@esp8266，可以编译成功，然后程序烧录到板子里能正常运行。

walker2048, 2018-12-28:
可以考虑用docker镜像哦，比WSL要好用，而且速度也快。[地址](https://cloud.docker.com/repository/registry-1.docker.io/walker2048/easy-aos)

zlasd, 2019-05-09:
出现以上warning的原因应该是windows和linux的换行符不同导致*.in文件解析失败，我在wsl中重新clone了一份代码之后解决了。

closed 2022-03-01 by skylarCai