alibaba/AliOS-Things #776: 调用linkkit_get_value();内存无法释放

xiaoyang432, 2018-12-25:
调用：
    linkkit_get_value(linkkit_method_get_service_input_value, thing, identifier,
                      NULL, &transparency_value);
的时候发现一个重大问题：
调试ESP8266的时候，从云端每发送一次数据，内存会逐步减少，感觉内存无法释放，屏蔽该函数问题解决。这个问题很严重，希望能够尽快解决。
现象如下所示：
[055070]<V> ============free heap size =12992==========

[060070]<V> ============free heap size =11776==========

[065070]<V> ============free heap size =10512==========
[070070]<V> ============free heap size =10512==========

[075070]<V> ============free heap size =9296==========

[080070]<V> ============free heap size =8080==========

**Screenshots**

**Additional context**

xiaowenxia, 2018-12-25:
@xiaoyang432 能具体描述一下问题吗？

xiaoyang432, 2018-12-25:
是例程linkkitapp，master

    linkkit_ops_t linkkit_ops = {
        .on_connect       = on_connect,       /* connect handler */
        .on_disconnect    = on_disconnect,    /* disconnect handler */
        .raw_data_arrived = raw_data_arrived, /* receive raw data handler */
        .thing_create     = thing_create,     /* thing created handler */
        .thing_enable     = thing_enable,     /* thing enabled handler */
        .thing_disable    = thing_disable,    /* thing disabled handler */
        .thing_call_service =
       thing_call_service, /* self-defined service handler */
        .thing_prop_changed = thing_prop_changed, /* property set handler */
        .linkit_data_arrived =
        linkit_data_arrived, /* transparent transmission data handler */
    };

xiaowenxia, 2018-12-25:
好的，谢谢你的问题，稍后回复一下你。

secretmikut, 2018-12-27:
针对你的case，linkkit_get_value有两种用法：

1. 
char *transparency_value = NULL;
linkkit_get_value(linkkit_method_get_service_input_value, thing, identifier, NULL, &transparency_value);
HAL_Free(transparency_value);

这种用法的情况下，需要自行释放内存

2.
int transparency_value = 0;
linkkit_get_value(linkkit_method_get_service_input_value, thing, identifier, &transparency_value, NULL);

这种情况下，直接获取int型数据的值，无需释放内存

closed 2022-03-01 by skylarCai