alibaba/AliOS-Things #777: ubuntu18.04下运行linkkit_gateway示例程序出错

wupanhao, 2018-12-26:
**Describe the bug**
按照教程搭建好aos开发环境编译出在linuxhost上运行的linkkit_gateway之后，可以正常运行，但是每当使用智能家居app扫码绑定的时候就会出错，看错误提示有点像内存问题？不知道有没有人遇到过类似的问题

**environment**
1. master分支: 
2. Ubuntu 18.04:
3. linkkit_gateway: 

**Expected behavior**
接入智能家居app测试

**Screenshots**
[image]

**Additional context**
```
[dbg] CoAPMessage_handle(753): --------Receive a Message------
[dbg] CoAPMessage_dump(422): *********Message Info**********
[dbg] CoAPMessage_dump(423): Version     : 1
[dbg] CoAPMessage_dump(424): Code        : 0.01(0x1)
[dbg] CoAPMessage_dump(425): Type        : 0x1
[dbg] CoAPMessage_dump(426): Msgid       : 2
[dbg] CoAPMessage_dump(427): Option      : 5
[dbg] CoAPMessage_dump(428): Payload Len : 66
[dbg] CoAPMessageToken_dump(405): Token Len   : 4
[dbg] CoAPMessageToken_dump(406): Token       : 9920A230
[dbg] CoAPMessage_dump(431): Remote      : 192.168.123.96:13629
[dbg] CoAPMessage_dump(438): ********************************
[inf] CoAPRequestMessage_handle(697): CoAPRequestMessage_handle:0x8169f28
[dbg] CoAPRequestMessage_handle(710): Request path is /dev/core/service/dev
[inf] CoAPResourceByPath_get(167): CoAPResourceByPath_get, context:0x8169f28
[dbg] CoAPResourceByPath_get(175): Found the resource: /dev/core/service/dev
[inf] recv_msg_handler(138): recv_msg_handler call callback
[inf] iotx_alcs_coap_ack_send(99): Send Ack Response Message: 2
[dbg] CoAPMessage_send(487): The message 2 isn't CON msg, needless to be retransmitted
[dbg] CoAPMessage_dump(422): *********Message Info**********
[dbg] CoAPMessage_dump(423): Version     : 1
[dbg] CoAPMessage_dump(424): Code        : 0.00(0x0)
[dbg] CoAPMessage_dump(425): Type        : 0x2
[dbg] CoAPMessage_dump(426): Msgid       : 2
[dbg] CoAPMessage_dump(427): Option      : 0
[dbg] CoAPMessage_dump(428): Payload Len : 0
[dbg] CoAPMessageToken_dump(405): Token Len   : 0
[dbg] CoAPMessageToken_dump(406): Token       : 
[dbg] CoAPMessage_dump(431): Remote      : 192.168.123.96:13629
[dbg] CoAPMessage_dump(438): ********************************
[inf] _alcs_event_handler(48): event_id 2
[inf] _alcs_event_handler(115): message response arrived: token=� �0, uri=/dev/core/service/dev, topic_msg={"id":"1","version":"1.0","params":{},"method":"core.service.dev"}
[inf] iotx_cm_local_conn_response_callback(350): rsp_type 11
[inf] iotx_cm_local_conn_process(1305): process 19
[inf] iotx_cm_local_conn_response_handler(808): URI = /dev/core/service/dev
[inf] dm_disp_event_new_data_received_handler(1434): IOTX_CM_EVENT_NEW_DATA_RECEIVED
[inf] dm_cmw_topic_callback(19): DMGR TOPIC CALLBACK
[inf] dm_cmw_topic_callback(26): DMGR Receive Message: /dev/core/service/dev
[inf] dm_disp_thing_dev_core_service_dev(1192): /dev/core/service/dev
[inf] dm_msg_request_parse(165): Current Request Message ID: 1
[inf] dm_msg_request_parse(174): Current Request Message Version: 1.0
[inf] dm_msg_request_parse(184): Current Request Message Method: core.service.dev
[inf] dm_msg_request_parse(193): Current Request Message Params: {}
[dbg] _dm_mgr_search_dev_by_devid(56): Device Found, devid: 0
WARNING,memory maybe corrupt!!
kernel panic,err 1!
Stack trace:
0 ./out/linkkit_gateway@linuxhost/binary/linkkit_gateway@linuxhost.elf() [0x8060629] 
1 ./out/linkkit_gateway@linuxhost/binary/linkkit_gateway@linuxhost.elf() [0x80606b1] 
2 ./out/linkkit_gateway@linuxhost/binary/linkkit_gateway@linuxhost.elf() [0x804fbe8] 
3 ./out/linkkit_gateway@linuxhost/binary/linkkit_gateway@linuxhost.elf() [0x80506b0] 
4 ./out/linkkit_gateway@linuxhost/binary/linkkit_gateway@linuxhost.elf() [0x809de9f] 
5 ./out/linkkit_gateway@linuxhost/binary/linkkit_gateway@linuxhost.elf() [0x8092e79] 
6 ./out/linkkit_gateway@linuxhost/binary/linkkit_gateway@linuxhost.elf() [0x8090274] 
7 ./out/linkkit_gateway@linuxhost/binary/linkkit_gateway@linuxhost.elf() [0x80d121c] 
8 ./out/linkkit_gateway@linuxhost/binary/linkkit_gateway@linuxhost.elf() [0x80d6dba] 
9 ./out/linkkit_gateway@linuxhost/binary/linkkit_gateway@linuxhost.elf() [0x80d7194] 
linkkit_gateway@linuxhost.elf: platform/arch/linux/cpu_impl.c:173: cpu_spin_unlock: Assertion `ret == 0' failed.
[1]    5390 abort (core dumped)  ./out/linkkit_gateway@linuxhost/binary/linkkit_gateway@linuxhost.elf

```

xiaowenxia, 2018-12-26:
linkkit和配网功能没有在linuxhost上充分验证过哈，你要体验配网功能的话，最好是使用具体的设备。

wupanhao, 2018-12-26:
可以不需要配网，我们的设备运行了一个嵌入式arm Linux系统，已经通过以太网正常联网，能够接入物联网平台，只是现在需要通过智能家居app绑定设备做一些手机端的联动，现在在ubuntu18.04下测试绑定过程出现了一点问题，想请问一下还有没有别的示例程序可以实现和智能家居app的绑定和一些简单的信息上下行

xiaowenxia, 2018-12-26:
目前本项目的alios things主要是面向于mcu的，如果你想要在linux上做数据上云的话，请参考这个：https://github.com/aliyun/iotkit-embedded。

wupanhao, 2018-12-26:
之前已经试过iotkit的demo，数据上云没问题，都能跑通，目前主要是想接入智能家居app做一下手机端的控制，但是一直没有找到一份合适的参考文档，请问有没有linkkit C SDK实现与智能家居app绑定这方面的案例呢？

Bloodance, 2018-12-27:
可以用AliOS-Things的linkkitapp，在已经适配的开发板上运行。

closed 2022-03-01 by skylarCai