alibaba/AliOS-Things #779: esp32devkitc: kernel panic when working on BLE mesh demo

Devilliv, 2018-12-26:
**Describe the bug**
参考https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-SIG-BLE-mesh-Setup-Guides.zh
搭建BLE mesh环境，在手机端用 nrf mesh app进行provision，输入OOB后，esp32产生kernel panic 然后reboot。

**environment**
1. master 
2. Desktop OS windows
3. 乐鑫ESP32 Devkitc一块
4. 微雪NRF51822开发板一块

**Screenshots**
# kernel panic,err 1200!
assertion "0" failed: file "platform/mcu/esp32/aos/soc_impl.c", line 35, function: soc_err_proc
abort() was called at PC 0x401168ab on core 0

Backtrace: 0x40087fd8:0x3ffdf720 0x400880d7:0x3ffdf740 0x401168ab:0x3ffdf760 0x4008a7d1:0x3ffdf790 0x40083db5:0x3ffdf7b0 0x40086695:0x3ffdf7d0 0x4008aed9:0x3ffdf7f0 0x4008aeb6:0x3ffdf810 0x40085151:0x3ffdf830 0x40085376:0x3ffdf850 0x40085527:0x3ffdf870 0x4010ae51:0x3ffdf890 0x40106cfb:0x3ffdf8b0 0x40106d6a:0x3ffdf900 0x40106da1:0x3ffdf920 0x401085ec:0x3ffdf950 0x40116333:0x3ffdf990 0x40085db5:0x3ffdf9d0

Rebooting...

gofortime, 2018-12-27:
> **Describe the bug**
> 参考https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-SIG-BLE-mesh-Setup-Guides.zh
> 搭建BLE mesh环境，在手机端用 nrf mesh app进行provision，输入OOB后，esp32产生kernel panic 然后reboot。 [...]

请描述一下你使用的代码分支，出现问题的步骤，以及复现问题的方法。

另外，可以使用esp32的工具dump一下crash的调用栈，具体方法如下：
`# xtensa-esp32-elf-addr2line -f 0x40087fd8:0x3ffdf720 0x400880d7:0x3ffdf740 0x401168ab:0x3ffdf760 0x4008a7d1:0x3ffdf790 0x40083db5:0x3ffdf7b0 0x40086695:0x3ffdf7d0 0x4008aed9:0x3ffdf7f0 0x4008aeb6:0x3ffdf810 0x40085151:0x3ffdf830 0x40085376:0x3ffdf850 0x40085527:0x3ffdf870 0x4010ae51:0x3ffdf890 0x40106cfb:0x3ffdf8b0 0x40106d6a:0x3ffdf900 0x40106da1:0x3ffdf920 0x401085ec:0x3ffdf950 0x40116333:0x3ffdf990 0x40085db5:0x3ffdf9d0 -e <your_elf_file>`
请把你的dump信息贴在这里。

gofortime, 2019-01-15:
@Devilliv Any update on this issue?

Devilliv, 2019-03-01:
@gofortime 
sorry，暂未对问题作进一步分析，回退代码至2018-03-02（教程发布日期），即可正常工作，谢谢！

closed 2019-03-01 by Devilliv

essencelite, 2020-10-20:
我遇到了类似的问题，最后查出来是调用这个的task的stack overflow了。
原来是2048，简单测试修改为8192就解决了。