alibaba/AliOS-Things #787: esp8266 rel 2.0.0 日志输出乱码

Torchwoods, 2018-12-29:
**Describe the bug**
esp8266 rel 2.0.0 日志输出乱码，波特为921600， 
[image]

xiaowenxia, 2018-12-29:
这个情况很大可能性是串口工具的问题哦，你换一个串口工具，或者重启这个工具试一试。

Lucifer3502, 2019-01-07:
我的helloworld 编译烧录后直接崩溃。基本的demo都跑不通。无语了！！！！
[image]

Lucifer3502, 2019-01-07:
[image]
[image]

xiaowenxia, 2019-01-07:
@Lucifer3502  请问你是基于哪个分支的呢？能否把你的binary发出来，我这边验证一下？

Lucifer3502, 2019-01-07:
clone的master分支。然后checkout tag V2.0.0. 
1. 我定位到编译的时候，写死了编译选项，而不能自定义。截图为 mcu\esp8266\gen_crc_bin.mk 的部分内容。
[image]
2. 在gen_appbin.py文件中，读取编译选项内容，并生成bin。
[image]
3. 我通过修改参数，改成和我使用espressif原版的编译配置一样，1Mflash，QIO, 40M. 编译后运行结果一样。另外，修改了eagle.app.v6.new.1024.app1.ld文件的irom0_0_seg， len= 0x6b000， 还有gen_appbin.py文件data_bin = struct.pack('<BBBBI',BIN_MAGIC_IROM,4,0,app,long(entry_addr,16))。我不知道还有那个地方需要修改？ 
[image]

4. 附件是 使用默认编译选项生成的bin，并且打包成了zip格式。
[helloworld@esp8266.zip](https://github.com/alibaba/AliOS-Things/files/2732067/helloworld.esp8266.zip)

xiaowenxia, 2019-01-08:
@Lucifer3502 
我们这边只在2M flash以上的esp8266上面验证过。
你如果更改过参数，需要注意一下esp8266的烧录工具也要进行相应的设置。
你这固件用nodeMCU是能够正常运行的（DIO、40M、4Mflash）：

```

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

OS SDK ver: 2.0-dev(d03acf5) compiled @ Feb 23 2018 19:45:05
empty
Wifi init success!!
mode : softAP(5e:cf:7f:b1:1c:d2)
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100
trace config close!!!
[000050]<V> aos framework init.
[000060]<V> application started.
[WiFi] event 8
[WiFi] event 8
[001080]<V> helloworld app_delayed_action:9 main

[WiFi] event 8
[WiFi] event 8
[WiFi] event 8
[WiFi] event 8
[WiFi] event 8
[WiFi] event 8
[WiFi] event 8
[WiFi] event 8
[WiFi] event 8
[006080]<V> helloworld app_delayed_action:9 main

[WiFi] event 8
[WiFi] event 8
[WiFi] event 8
[WiFi] event 8
[WiFi] event 8
[WiFi] event 8
[WiFi] event 8
[011080]<V> helloworld app_delayed_action:9 main
```

sinboun, 2019-11-07:
[image]
也有类似的问题

sinboun, 2019-11-07:
@Torchwoods 最后解决了吗

closed 2022-03-01 by skylarCai