alibaba/AliOS-Things #793: AliOS things的LOG宏定义可能有点问题

prowan, 2019-01-04:
**Describe the bug**

**environment**
1. branch(version): 
2. Desktop OS(windows linux or macOS):
3. example: 
#include <aos/aos.h>

int i = 0;
int k = 0;
static void app_delayed_action(void *arg)
{
    LOG("fff  %s:%d %s %d     %d\r\n", __func__, __LINE__, aos_task_name(), i, k++);
    i++;
    
    aos_post_delayed_action(500, app_delayed_action, NULL);
}
 
int application_start(int argc, char *argv[])
{
    LOG("application started.");
    aos_post_delayed_action(1000, app_delayed_action, NULL);
    aos_loop_run();

    return 0;
}

**Expected behavior**
[001620]<V> fff  app_delayed_action:11 main 1     3

[002140]<V> fff  app_delayed_action:11 main 2     5

[002660]<V> fff  app_delayed_action:11 main 3     7

[003180]<V> fff  app_delayed_action:11 main 4     9

[003700]<V> fff  app_delayed_action:11 main 5   11

[004220]<V> fff  app_delayed_action:11 main 6   13

[004740]<V> fff  app_delayed_action:11 main 7     15

**Screenshots**
[image]

[](url)
**Additional context**

xiaowenxia, 2019-01-04:
@prowan 请问你是在哪个开发板上面运行的呢？

xiaowenxia, 2019-01-04:
@prowan 谢谢你的问题，目前LOG系统里面是有进行多次调用参数的情况，导致`k++`实际上运行了2次：
```c
#define log_print(CON, MOD, COLOR, LVL, ...) \
    do { \
        if (CON) { \
            csp_printf("[%06d]<" LVL "> ", (unsigned)aos_now_ms(), __VA_ARGS__); \
            csp_printf(__VA_ARGS__); \
            csp_printf("\r\n"); \
        } \
    } while (0)
```
后面会进行相应的改善。
不过请最好不要在LOG里面做过多的逻辑运算哈。

closed 2022-03-01 by skylarCai