alibaba/AliOS-Things #806: The example of pca10040 in alios things can't run

linyuanding, 2019-01-12:
**Describe the bug**
The routine of the development board pca10040 of mcu nrf52832 can't run, but the wiki's documentation says it can run. I hope someone can help me solve this problem.

**environment**
1. branch(version): 
2. Desktop OS(windows linux or macOS):windows
3. example: 
The power management routines mentioned in the wiki documentation are based on the pca10040 development board, and I burned this routine without seeing the exmple mentioned in the documentation.
**Expected behavior**
I hope someone can help me solve this problem.

**Screenshots**
[image]
LOG did not print any information after burning the sample program for power management

**Additional context**

xiaowenxia, 2019-01-12:
can you provide the address of the wiki document?   and  how did you burn the binary?

linyuanding, 2019-01-12:
Hello, the address in the wiki file is
https://github.com/alibaba/AliOS-Things/wiki/AliOS-Things-power-manage-framework, I used JLink to burn the binary?, the burning address is 0x00010000, I don't know if the download method is correct.

linyuanding, 2019-01-12:
Will you run the routine that successfully passed pca10040?

xiaowenxia, 2019-01-12:
maybe you need to burning the  bootloader.

linyuanding, 2019-01-12:
What you mean is that I should not burn the application but the bootloader？

linyuanding, 2019-01-13:
I did not burn the bootloader program, because I did not burn the bootloader when I ran the SDK provided by NORDIC. You reminded me of this, I went to try to burn the bootloader program.

linyuanding, 2019-01-13:
Hey, I checked out today that nrf52832 only has a bootloader about dfu, but I feel that this should not be too much related to my routine.

xiaowenxia, 2019-01-14:
@linyuanding the bootloader sample code is here: https://github.com/alibaba/AliOS-Things/tree/master/platform/mcu/nrf52xxx/bootloader  you can compile & flash into 0x00 address.

linyuanding, 2019-01-14:
I will give it a try.

linyuanding, 2019-01-14:
This bootloader demo lacks file compilation and does not pass, and this seems to be about the pca10056 development board OTA bootloader.

xiaowenxia, 2019-01-14:
we used `pc-nrfutil`(provide by nordic) to generate bootloader.hex and settings.hex, you can try this: https://github.com/NordicSemiconductor/pc-nrfutil

linyuanding, 2019-01-14:
Ok, let me try it, I always feel that it is not the reason for the bootloader.

xiaowenxia, 2019-01-14:
use command:  `aos upload pwr_test@pca10040`, or use AliOS-Studio:  

[image]

linyuanding, 2019-01-14:
Can you show me the log?

xiaowenxia, 2019-01-14:
but first, you need to add JLink directory into PATH environment，for more information: https://dev.iot.aliyun.com/post/096848?spm=a2c56.12366940.0.0.7eb25206QT077q

this is my `aos upload` log:

```
win10:/d/workspace/github/xiaowenxia/AliOS-Things$ aos upload helloworld@pca10040
aos-cube version: 0.2.64
[INFO]: Target: helloworld@pca10040
[INFO]: Currently in aos_sdk_path: 'D:\workspace\github\xiaowenxia\AliOS-Things'
scons: Reading SConscript files ...
[INFO]: Scons to upload!args:{'APPLICATION': 'helloworld', 'COMMAND': 'upload', 'BOARD': 'pca10040'}

[INFO]: Target: helloworld@pca10040
[INFO]: Currently in aos_sdk_path: 'D:\workspace\github\xiaowenxia\AliOS-Things'

[INFO]: Running cmd:
        'python D:\workspace\github\xiaowenxia\AliOS-Things/build/site_scons/jlink.py -d nRF52840_xxAA -i swd -f D:\workspace\github\xiaowenxia\AliOS-Things/out/helloworld@pca10040/binary/helloworld@pca10040.bin -p 0x00010000'
SEGGER J-Link Commander V6.22a (Compiled Nov 28 2017 17:56:48)
DLL version V6.40, compiled Oct 26 2018 15:06:02

Script file read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Jul 12 2018 11:44:41
Hardware version: V1.00
S/N: 682552772
VTref = 3.300V
Target connection not established yet but required for command.
Device "NRF52840_XXAA" selected.

Connecting to target via SWD
Found SW-DP with ID 0x2BA01477
Found SW-DP with ID 0x2BA01477
Scanning AP map to find all available APs
AP[2]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x24770011)
AP[1]: JTAG-AP (IDR: 0x02880000)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
Found Cortex-M4 r0p1, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 003BB002 DWT
ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 002BB003 FPB
ROMTbl[0][3]: E0000000, CID: B105E00D, PID: 003BB001 ITM
ROMTbl[0][4]: E0040000, CID: B105900D, PID: 000BB9A1 TPIU
ROMTbl[0][5]: E0041000, CID: B105900D, PID: 000BB925 ETM
Cortex-M4 identified.
PC = 00011FCA, CycleCnt = 000EC81E
R0 = 00000001, R1 = 00000001, R2 = 00000003, R3 = 00011FC5
R4 = 00000001, R5 = 20001434, R6 = 2000185C, R7 = 20001980
R8 = 00000001, R9 = 2000187C, R10= 200018FC, R11= 0000074A
R12= 00000000
SP(R13)= 20001E40, MSP= 20006038, PSP= 20001E40, R14(LR) = 000135CB
XPSR = 61000000: APSR = nZCvq, EPSR = 01000000, IPSR = 000 (NoException)
CFBP = 02000001, CONTROL = 02, FAULTMASK = 00, BASEPRI = 00, PRIMASK = 01

FPS0 = 00000000, FPS1 = 00000000, FPS2 = 00000000, FPS3 = 00000000
FPS4 = 00000000, FPS5 = 00000000, FPS6 = 00000000, FPS7 = 00000000
FPS8 = 00000000, FPS9 = 00000000, FPS10= 00000000, FPS11= 00000000
FPS12= 00000000, FPS13= 00000000, FPS14= 00000000, FPS15= 00000000
FPS16= 00000000, FPS17= 00000000, FPS18= 00000000, FPS19= 00000000
FPS20= 00000000, FPS21= 00000000, FPS22= 00000000, FPS23= 00000000
FPS24= 00000000, FPS25= 00000000, FPS26= 00000000, FPS27= 00000000
FPS28= 00000000, FPS29= 00000000, FPS30= 00000000, FPS31= 00000000
FPSCR= 00000000

Downloading file [D:\workspace\github\xiaowenxia\AliOS-Things/out/helloworld@pca10040/binary/helloworld@pca10040.bin]...
J-Link: Flash download: Bank 0 @ 0x00000000: 1 range affected (110592 bytes)
J-Link: Flash download: Total time needed: 2.043s (Prepare: 0.052s, Compare: 0.010s, Erase: 0.080s, Program: 1.888s, Verify: 0.002s, Restore: 0.008s)
O.K.

Script processing completed.

set device: nRF52840_xxAA
set interface: swd
set loadfile: D:\workspace\github\xiaowenxia\AliOS-Things/out/helloworld@pca10040/binary/helloworld@pca10040.bin
set address: 0x00010000
host os: Windows
gen jlink commands file: D:\workspace\github\xiaowenxia\AliOS-Things\out\helloworld@pca10040\binary\nRF52840_xxAA.jlink
execute cmd: JLink.exe -device nRF52840_xxAA -if swd -speed 4000k -CommanderScript D:\workspace\github\xiaowenxia\AliOS-Things\out\helloworld@pca10040\binary\nRF52840_xxAA.jlink
---host_os:Win32
[INFO]: Firmware upload succeed!

scons: done reading SConscript files.
scons: Building targets ...
scons: `.' is up to date.
scons: done building targets.

```

linyuanding, 2019-01-14:
I mean, after the burning is successful, the program runs the log printed by the serial port.

xiaowenxia, 2019-01-14:
```
# [000000]<V> pwr test started.
count1 = 0, idle = 0
count2 = 0
count1 = 1, idle = 2
count1 = 2, idle = 3
count2 = 1
count1 = 3, idle = 4
count1 = 4, idle = 5
count2 = 2
count1 = 5, idle = 6
count1 = 6, idle = 7
count2 = 3
count1 = 7, idle = 8
count1 = 8, idle = 9
count2 = 4
count1 = 9, idle = 10
count1 = 10, idle = 11
count2 = 5
count1 = 11, idle = 12
count1 = 12, idle = 13
count2 = 6

...
```

linyuanding, 2019-01-14:
thank! ,let me try again.

linyuanding, 2019-01-14:
Sorry, my program burned in or did not respond, the serial port tool can not see the log, you help me to see the download log information is correct?

> Executing task: aos upload helloworld@pca10040 <

aos-cube version: 0.2.64
[INFO]: Target: helloworld@pca10040
[INFO]: Currently in aos_sdk_path: 'D:\AliOS-Things'

scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly
File "build\site_scons\site_init.py", line 39, in aos_global_config
scons: Reading SConscript files ...
[INFO]: Scons to upload!args:{'APPLICATION': 'helloworld', 'COMMAND': 'upload', 'BOARD': 'pca10040'}

[INFO]: Target: helloworld@pca10040
[INFO]: Currently in aos_sdk_path: 'D:\AliOS-Things'

[INFO]: Running cmd:
        'python D:\AliOS-Things/build/site_scons/jlink.py -d nRF52832_xxAA -i swd -f D:\AliOS-Things/out/helloworld@pca10040/binary/helloworld@pca10040.bin -p
0x00010000'
set device: nRF52832_xxAA
set interface: swd
set loadfile: D:\AliOS-Things/out/helloworld@pca10040/binary/helloworld@pca10040.bin
set address: 0x00010000
host os: Windows
gen jlink commands file: D:\AliOS-Things\out\helloworld@pca10040\binary\nRF52832_xxAA.jlink
execute cmd: JLink.exe -device nRF52832_xxAA -if swd -speed 4000k -CommanderScript D:\AliOS-Things\out\helloworld@pca10040\binary\nRF52832_xxAA.jlink
SEGGER J-Link Commander V6.20c (Compiled Sep 26 2017 17:31:29)
DLL version V6.20c, compiled Sep 26 2017 17:30:56

Script file read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Jul 12 2018 11:44:41
Hardware version: V1.00
S/N: 682898873
VTref = 3.300V
Target connection not established yet but required for command.
Device "NRF52832_XXAA" selected.

Connecting to target via SWD
Found SW-DP with ID 0x2BA01477
Found SW-DP with ID 0x2BA01477
Scanning AP map to find all available APs
AP[2]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x24770011)
AP[1]: JTAG-AP (IDR: 0x02880000)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
Found Cortex-M4 r0p1, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS
ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 003BB002 DWT
ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 002BB003 FPB
ROMTbl[0][3]: E0000000, CID: B105E00D, PID: 003BB001 ITM
ROMTbl[0][4]: E0040000, CID: B105900D, PID: 000BB9A1 TPIU
ROMTbl[0][5]: E0041000, CID: B105900D, PID: 000BB925 ETM
Cortex-M4 identified.
PC = FFFFFFFE, CycleCnt = 00000000
R0 = 00000000, R1 = 00000000, R2 = 00000000, R3 = 00000000
R4 = 00000000, R5 = 00000000, R6 = 00000000, R7 = 00000000
R8 = 00000000, R9 = 00000000, R10= 00000000, R11= 00000000
R12= 00000000
SP(R13)= FFFFFFD8, MSP= FFFFFFD8, PSP= 00000000, R14(LR) = FFFFFFF9
XPSR = 01000003: APSR = nzcvq, EPSR = 01000000, IPSR = 003 (HardFaultMemManage)
CFBP = 00000000, CONTROL = 00, FAULTMASK = 00, BASEPRI = 00, PRIMASK = 00
FPU regs: FPU not enabled / not implemented on connected CPU.

Downloading file [D:\AliOS-Things/out/helloworld@pca10040/binary/helloworld@pca10040.bin]...
J-Link: Flash download: Bank 0 @ 0x00000000: 1 range affected (110592 bytes)
J-Link: Flash download: Total time needed: 2.113s (Prepare: 0.049s, Compare: 0.010s, Erase: 0.080s, Program: 1.962s, Verify: 0.002s, Restore: 0.008s)
O.K.

Script processing completed.

---host_os:Win32
[INFO]: Firmware upload succeed!

scons: done reading SConscript files.
scons: Building targets ...
scons: `.' is up to date.
scons: done building targets.

终端将被任务重用，按任意键关闭。

xiaowenxia, 2019-01-14:
yes,

linyuanding, 2019-01-14:
Can you send me a copy of your bootloader, I found a lot of bootloader here do not know which burn?

xiaowenxia, 2019-01-14:
Got an email ? I send via email.

linyuanding, 2019-01-14:
yes！，linyuanxun@lmengchian.com

linyuanding, 2019-01-14:
Made it? @xiaowenxia

xiaowenxia, 2019-01-14:
@linyuanding I can't,  DNS failed。I generated download link：

* [nrf52832_xxaa.hex](https://space.dingtalk.com/s/gwHOADtEwwLOBJYRigPaACA1OWJmZjlkOWI4Yjk0MzNhYTQ3NTYwMGU5OGI3MWYxNg) 密码: QnMw

* [settings_single.hex](https://space.dingtalk.com/s/gwHOADtExALOBJYRigPaACBiMzdkZjk0ZDZjMjM0N2RiYWU3ZDNjNzg2YThiZjA2OQ) 密码: uKB5

note:

* nrf52832_xxaa.hex   flash to 0x00000000 
* settings_single.hex flash to 0x0007F000 

flash partition described here： https://github.com/alibaba/AliOS-Things/blob/master/platform/mcu/nrf52xxx/nrf52832_xxaa.ld

linyuanding, 2019-01-14:
The program is running, thank you very much! ! !

xiaowenxia, 2019-01-14:
nice work!!!  😄

closed 2019-01-14 by xiaowenxia