alibaba/AliOS-Things #812: 使用AliOS-Things-developer/example/linkdevelop/device_ctrl的例程，修改了三元组烧录到develop kit开发板中，云端（飞燕）设备没有显示在线

YeFly9, 2019-01-14:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Additional context**

xiaowenxia, 2019-01-14:
@YeFly9 你把设备端的运行log发上来看看哈。

YeFly9, 2019-01-14:
[image]

xiaowenxia, 2019-01-14:
@YeFly9 是运行的log哈。确认一下，你在设备端有输入netmgr connect <BSSID> <passwd> 把developerkit连上wifi吗？

YeFly9, 2019-01-14:
输入netmgr connect没有操作，能具体说一下怎么操作吗？

YeFly9, 2019-01-14:
[image]

YeFly9, 2019-01-14:
我的使用的是git bush，VSCode不能使用串口，上面是我用串口调试助手打印的

xiaowenxia, 2019-01-14:
[image]

xiaowenxia, 2019-01-14:
[image]

YeFly9, 2019-01-14:
设备端怎么输入netmgr connect 把developerkit连上wifi呀？

xiaowenxia, 2019-01-14:
在串口工具上输入呀，注意一下是否使用`\r\n` 或者 `\n` 。或者你在cmd/ powershell下运行`aos monitor` 也可以输入。

YeFly9, 2019-01-14:
[image]

YeFly9, 2019-01-14:
串口输入没反应啊

xiaowenxia, 2019-01-14:
后面你加上wifi名称和wifi密码：

```
netmgr connect <wifi-name> <wifi-passwd>
```

YeFly9, 2019-01-14:
使用cmd/powershell下也不行
[image]

YeFly9, 2019-01-14:
这个是云端的设备日志
[image]

xiaowenxia, 2019-01-14:
@YeFly9 看起来是wifi模组不稳定，重启板子还会不会有问题？

YeFly9, 2019-01-15:
云端设备上线是不是您这边帮我操作的时候上线的，我从昨天试到现在设备再也没有上线了

xiaowenxia, 2019-01-15:
你把设备端的log保存成txt发给这里面哈，我看看。

YeFly9, 2019-01-15:
[111.txt](https://github.com/alibaba/AliOS-Things/files/2758183/111.txt)

xiaowenxia, 2019-01-15:
@YeFly9 看log里面，你的wifi为什么会循环的`Got IP`呢？

```
[233610]<I> Got ip : 192.168.2.112, gw : 192.168.2.1, mask : 255.255.255.0
[233630]<I> AT command AT+WMAC? succeed, rsp: 
+WMAC:e47deb1a6423
OK
[233630]<I> mac result: e47deb1a6423
[263080]<I> AT command AT+WMAC? succeed, rsp: 
+WMAC:e47deb1a6423
OK
[263090]<I> mac result: e47deb1a6423
[263090]<I> Got ip : 192.168.2.112, gw : 192.168.2.1, mask : 255.255.255.0

[263110]<I> AT command AT+WMAC? succeed, rsp: 
+WMAC:e47deb1a6423
OK
[263120]<I> mac result: e47deb1a6423
[330060]<I> AT command AT+WMAC? succeed, rsp: 
+WMAC:e47deb1a6423

OK
```

换一个wifi热点试试，直接用手机热点也行。

YeFly9, 2019-01-15:
好的，我试试

xiaowenxia, 2019-01-15:
你在串口里面输入help就能看的netmgr的使用说明哈，  `netmgr clear` 就能清除保存下来的wifi信息哈。

YeFly9, 2019-01-15:
大佬，我现在设备复位能看到上线过，但是设备不是一直在线的，这是串口打印的信息
[222.txt](https://github.com/alibaba/AliOS-Things/files/2758637/222.txt)

xiaowenxia, 2019-01-15:
刚才确认了一下，是云端主动断开连接了，有可能是限流，或者三元组错误，或者三元组冲突（重复登录）等。

YeFly9, 2019-01-15:
我重新上电了一下，莫名奇妙的上线了

closed 2022-03-01 by skylarCai