alibaba/AliOS-Things #828: Bluetooth breeze component related issues

linyuanding, 2019-01-16:
**Is your feature request related to a problem? Please describe.**
I am now using the breeze component to dock with the APP of the Feiyan platform, but I found that there are too few interfaces in the breeze, and the implementation of the interface is also packaged into a library. I did not find the interface to modify the Bluetooth name, nor found it. I found the interface to the data, do I need to call the underlying library?

**Describe the solution you'd like**
There are fewer Bluetooth development documents, so I hope someone tells me how to do it.

**Additional context**

luxuguang-leo, 2019-01-17:
Hi Linyuanding,

Thanks for your response.

- Before rel2.0.0 version, breeze component is delivered in the form of library, and hopefully will be open source in further version release.

- For user's APIs, please refer to https://github.com/alibaba/AliOS-Things/wiki/%E8%93%9D%E7%89%99%E8%BE%85%E5%8A%A9WiFi%E9%85%8D%E7%BD%91HAL%E8%AF%B4%E6%98%8E , the APIs is  the same as "蓝牙协议栈HAL列表".

- For bluetooth name, the default name in the advertising packet is "AZ" which is filled within， but you can still advertise the full name in the scan response packet. Related code snippet:
``
ais_err_t ble_advertising_start(ais_adv_init_t *adv)
{
    int            err;
    uint8_t        flag = 0, dis[] = { 0x0a, 0x18 }, ad_len = 3, sd_len = 1;
    struct bt_data ad[ad_len];
    struct bt_data sd[sd_len];
    //...
     switch (adv->name.ntype) {
        case AIS_ADV_NAME_SHORT:
            sd[0].type = BT_DATA_NAME_SHORTENED;
            break;
        case AIS_ADV_NAME_FULL:
            sd[0].type = BT_DATA_NAME_COMPLETE;
            break;
        default:
            printf("Invalid adv name type.\r\n");
            return AIS_ERR_INVALID_ADV_DATA;
    }

    if (adv->name.name == NULL) {
        printf("Invalid adv device name.\r\n");
        return AIS_ERR_INVALID_ADV_DATA;
    }

    sd[0].data     = adv->name.name;
    sd[0].data_len = strlen(adv->name.name);
``

Just replace sd data and change the adv name in the scan response pkt within.

Thanks,
Leo

linyuanding, 2019-01-17:
@luxuguang-leo 
Hi,luxuguang-leo
I want to ask you another question. Can my device send ble_send_notification or ble_send_indication function to send data to the APP? What needs attention?

[image]

luxuguang-leo, 2019-01-17:
> @luxuguang-leo
> Hi,luxuguang-leo
> I want to ask you another question. Can my device send ble_send_notification or ble_send_indication function to send data to the APP? What needs attention?
> 
> [image]

Hi Linyuanding,

Of course you can, the two APIs are called in Breeze components, and after breeze connection is established, you can directly call applications  APIs like `breeze_post` `breeze_post_fast` 'breeze_post_ext' which are 
 defined in [https://github.com/alibaba/AliOS-Things/wiki/Breeze-%E7%94%A8%E6%88%B7%E7%BC%96%E7%A8%8B%E6%8E%A5%E5%8F%A3%E8%AF%B4%E6%98%8E](url)

Hope that helps.
Thanks,
Leo

linyuanding, 2019-01-17:
Sorry, I didn't open the link you sent me.

[image]

luxuguang-leo, 2019-01-17:
> Sorry, I didn't open the link you sent me.
> 
> [image]

Sorry for the invalid links:

luxuguang-leo, 2019-01-17:
> Sorry, I didn't open the link you sent me.
> 
> [image]

Correct the links:https://github.com/alibaba/AliOS-Things/wiki/Breeze-%E7%94%A8%E6%88%B7%E7%BC%96%E7%A8%8B%E6%8E%A5%E5%8F%A3%E8%AF%B4%E6%98%8E

linyuanding, 2019-01-17:
thank you very much！！！

linyuanding, 2019-01-17:
Hi  @luxuguang-leo 
I will report an error when calling breeze_post，However, when I searched according to this error path, I could not find the file he had reported.

[image]

luxuguang-leo, 2019-01-17:
> when

Hi Linyuanding,

Which branch are you using? Please get the latest breeze update on rel2.0.0. 
Besides,  before you call breeze_post, make sure the breeze security connection established, or you should not call it.

Thanks.
Leo

linyuanding, 2019-01-19:
Thank you, but when I use the breeze_post function to send a message, I found that I can only send a message if I call it in the callback function that receives the message.

closed 2019-01-22 by linyuanding