alibaba/AliOS-Things #841: 蓝牙例程breeze无法正常发送数据

linyuanding, 2019-01-19:
我只能在接收数据的回调函数中才能把数据APP才能收到消息，我自己写一个定时器去发消息APP就收不到消息了。
希望有人能告诉我为什么在接收数据的回调中能发送数据，在回调以外就不能发送了？

luxuguang-leo, 2019-01-22:
Hi，
直接调用breeze_post是OK的，但你需要保证发消息不block yloop事件，因为breeze组件依赖底层的yloop机制，如果在上层使用while循环会阻碍yloop机制，推荐使用方式：

main{
aos_post_delayed_action( , simulate_data, NULL);
aos_loop_run();
}

void simuate_data()
{
    //
    //breeze_post();
   //aos_post_delayed_action(, simulate_data, NULL)
}

linyuanding, 2019-01-23:
感谢，我试试这个方法。

linyuanding, 2019-01-23:
嗨！
我用了你说的方法去发送数据，还是不行 
[image]
[image]

luxuguang-leo, 2019-01-23:
发送数据的判断条件应该是认证通道结束之后，而不是蓝牙连接

linyuanding, 2019-01-23:
按照你说的，我把判断条件了为了，认证成功之后再去发送，依然没成功。
[image]

linyuanding, 2019-01-23:
没有错误信息啊，log还是跟之前一样，breeze_post（）调用完成后触发dev_status_changed_handler（），打印“dev_status_changed(): Tx-done.”表示发送完成
[image]

linyuanding, 2019-01-23:
[image]

luxuguang-leo, 2019-01-23:
这个demo的APP的逻辑应该是在收到非自己发送的数据处理后不会打印，显示Tx-done应该是数据通过indication发送成功了。

linyuanding, 2019-01-23:
你的意思是我的数据已经发送过去了，APP没有打印出来？

luxuguang-leo, 2019-01-23:
你可以抓包或者在协议栈加打印来确认indication有无发送成功

linyuanding, 2019-01-23:
我在底层打印显示成功，我买个抓包器试试。

closed 2022-03-01 by skylarCai