alibaba/AliOS-Things #848: Linkkitapp@esp8266可以配网成功，但是APP绑定失败

mq9494, 2019-01-25:
使用APP进行配网，可以成功配网，在云端调试也能看到设备在线，但是点击APP的开始使用，超时并且提示绑定失败。不知道问题在哪里。云端可以看到LinkkitSDK版本2.2.1。

xiaowenxia, 2019-01-25:
@mq9494 esp8266 需要按相应的按键进入到配网模式的，你确认一下哈。

mq9494, 2019-01-25:
确认进入配网模式，并且配网成功。
[image]
到这一步卡住了。

kid2682, 2019-01-25:
@mq9494 麻烦提供一下设备端的日志，和测试版app的日志。 需要重新下载一个测试版的app

mq9494, 2019-01-25:
@kid2682 请帮忙看一下，谢谢啦～
[2019-01-25-13-02.txt](https://github.com/alibaba/AliOS-Things/files/2794713/2019-01-25-13-02.txt)
[linkkit@esp8266_log.txt](https://github.com/alibaba/AliOS-Things/files/2794714/linkkit%40esp8266_log.txt)

kid2682, 2019-01-25:
@mq9494 是基于alios的还是某个操作系统+iotkit sdk。如果是纯sdk，可以从这里下载最新版本的。
https://github.com/aliyun/iotkit-embedded   v2.3.0是最新的，v2.2.1里面修复了一些bug

mq9494, 2019-01-25:
@kid2682 是基于alios的。是直接把V2.3.0替换掉V2.2.1吗。/middleware/linkkit/sdk-c
[image]

kid2682, 2019-01-25:
@mq9494 alios things是master？ 切换到rel_2.0.0试试，有问题的话请抓一个设备端的日志给我。

mq9494, 2019-01-25:
@kid2682 好的，谢谢。我切换之后再试一下。

mq9494, 2019-01-25:
@kid2682 你好，这个是什么意思
“发现设备时其 AuthorityEntity 实例不存在，只有在纯本地环境才创建 IMSThing”

mq9494, 2019-01-25:
@kid2682 你好,这个是什么意思？
“发现设备时其 AuthorityEntity 实例不存在，只有在纯本地环境才创建 IMSThing”

mq9494, 2019-01-25:
@kid2682 你好，这个是rel_2.0.0设备端的日志
[linkkit@esp8266_log1520.txt](https://github.com/alibaba/AliOS-Things/files/2795255/linkkit%40esp8266_log1520.txt)
问题还是绑定失败。不知道怎么办
这个是APP的日志
[2019-01-25-16-30.txt](https://github.com/alibaba/AliOS-Things/files/2795258/2019-01-25-16-30.txt)

mq9494, 2019-01-25:
@kid2682 有的，每次编译前都clean。
是不是我的编译环境或者步骤有问题呀？
Ubuntu 18.04，使用code。linkkitapp中我自己修改了波特率（hal_uart_init()），然后在example_solo.c中注释掉了循环中的那几个post，在property_changed_handler增加了things/service/property/set的LightSwitch。
然后aos make clean，aos make linkkitapp@esp8266。
下载的四个.bin 文件地址看网上的是0x0000，0x1000, 0x3fc000, 0x3fe000。

kid2682, 2019-01-25:
应该没多大关系，只是觉得你的log有点奇怪。我再看一下。

mq9494, 2019-01-25:
@kid2682 好的，麻烦您了。刚接触这个，还不太懂

mq9494, 2019-01-26:
@kid2682 是不是因为我用的是调试设备的测试设备的三元组。还是说需要购买设备激活码？

fenghr1219, 2019-01-28:
@mq9494 您这里的APP是开发版（测试版）的智能家居还是正式版的智能家居呢？

mq9494, 2019-01-31:
@fenghr1219 你好，昨天重新clone，使用rel_2.0.0之后，可以绑定成功了。忘记反馈了。谢谢您还关心

mq9494, 2019-01-31:
@fenghr1219 之前的日志，设备端和APP端的时间基本都是同步的。

closed 2019-01-31 by mq9494

fenghr1219, 2019-01-31:
@mq9494 好的 您这里问题解决了就好