alibaba/AliOS-Things #849: DEBUG时出现找不到debugger的情况

yayayi333, 2019-01-28:
[image]

按照提示进行开发环境配置，在启动DEBUG时，出现如图提示信息。请问是什么原因？

xiaowenxia, 2019-01-28:
你的`MIDebuggerPath`是怎么写的呢？

yayayi333, 2019-01-28:
"customLaunchSetupCommands": [],
            "launchCompleteCommand": "exec-run",
            "osx": {
                "MIMode": "gdb",
                "miDebuggerPath": "arm-none-eabi-gdb"
            },
            "linux": {
                "MIMode": "gdb",
                "miDebuggerPath": "arm-none-eabi-gdb"
            },
            "windows": {
                "MIMode": "gdb",
                "miDebuggerPath": "arm-none-eabi-gdb.exe"
            }

xiaowenxia, 2019-01-28:
`arm-none-eabi-gdb.exe` 所在的目录是不是不在你的Path环境变量里面？ 终端里面看看能不能运行`arm-none-eabi-gdb.exe`.

yayayi333, 2019-01-28:
[image]
测试可以运行

yayayi333, 2019-01-28:
可以运行，然后出现新的报错，ST Utilty确认安装完成。
[image]

xiaowenxia, 2019-01-28:
这里面没有使用ST-Utilty哦，使用的是./build/cmd/win32/st-util.exe 这个软件，你确认下这个路径对不对。

xiaowenxia, 2019-01-28:
@yayayi333 AliOS-Things目录下面会生成`.gdbinit`，看看这里面的st-util.exe 配置的是否正确？

xiaowenxia, 2019-01-28:
@yayayi333  `aos make clean` 然后重新编译哦。

yayayi333, 2019-01-28:
直接修改.gdbinit中的路径是否可行？

xiaowenxia, 2019-01-28:
也可以的，但是`.gdbinit`是`aos make app@board`的时候自动生成的，下次编译的时候会自动更改.gdbinit.

xiaowenxia, 2019-01-28:
👍

closed 2019-01-28 by xiaowenxia