alibaba/AliOS-Things #850: linkkitapp esp8266 ota升级失败

nostalgia-w, 2019-01-30:
板子：esp8266，alios things 的master分支

进行固件升级的时候，从设备日志来看，程序是下载了，但是有error，且“智能生活开发平台”上显示固件升级失败，原因是 上报版本异常。

[2019-01-30_15:55:26][091050]<I> s:763429 512
[2019-01-30_15:55:26][091050]<I> s:763941 512
[2019-01-30_15:55:26][091060]<I> s:764453 512
[2019-01-30_15:55:26][091060]<I> s:764965 512
[2019-01-30_15:55:26][091060]<I> s:765477 512
[2019-01-30_15:55:26][091070]<I> s:765989 512
[2019-01-30_15:55:26][091070]<I> s:766501 512
[2019-01-30_15:55:26][091080]<I> s:767013 512
[2019-01-30_15:55:26][091090]<I> s:767525 512
[2019-01-30_15:55:26]E: 203 HAL_TCP_Read() | connection is closed
[2019-01-30_15:55:26][091090]<I> s:767908 383
[2019-01-30_15:55:26][091100]<I> url md5=20ADE91AF06A627911D2C3F6BCC80E41
[2019-01-30_15:55:26][091100]<I> digestMD5=20ADE91AF06A627911D2C3F6BCC80E41
[2019-01-30_15:55:26][091100]<I> bin_size = 767908
[2019-01-30_15:55:26][091100]<I> bin_size = 767908
[2019-01-30_15:55:26][091100]<I> off_set = 767880
[2019-01-30_15:55:26][091110]<I> ota_image_identity.image_magic=3ffe8c18
[2019-01-30_15:55:26][091110]<I> ota_image_identity.image_size=3fff1f00
[2019-01-30_15:55:26][091110]<I> ota_image_identity.image_md5_value=000000000000000000000000000000FC
[2019-01-30_15:55:26][091120]<I> ota_image_identity.image_crc16=68a7
[2019-01-30_15:55:26][091120]<E> image magic error
[2019-01-30_15:55:26][091130]<E> ota check image failed
[2019-01-30_15:55:26][091130]<I> reboot system after 3 second!
[2019-01-30_15:55:29][094130]<I> task update over
[2019-01-30_15:55:29]reboot!

huafeihua161, 2019-01-30:
从log看 你下载的固件内部的MD5校验未通过，正常生成的固件会在固件尾部增加固件信息 及md5值等

nostalgia-w, 2019-01-30:
下载带.md5.bin后缀的固件可以成功，谢谢！

之前用的rel1.3.1，烧录到板子里的和固件下载的都是直接.bin后缀的，所以以为现在这个master分支也是。

closed 2019-01-30 by xiaowenxia