alibaba/AliOS-Things #851: 构建tinyengie_app失败，目录文件缺失

chenyulun, 2019-01-30:
**Is your feature request related to a problem? Please describe.**
```
> Executing task: aos make tinyengine_app@developerkit <
.//build/aos_target_config.mk:334: *** Unknown component: middleware.linkkit.iotkit - directory or makefile for component not found. Ensure the middleware/linkkit/iotkit directory contains iotkit.mk.  Stop.
make: *** No rule to make target `out/tinyengine_app@developerkit/config.mk', needed by `main_app'.  Stop.
[AliOS-Things] ERROR: "/Users/mac/github/AliOS-Things/build/cmd/osx/make" returned error code 2.
[AliOS-Things] ERROR: Command "/Users/mac/github/AliOS-Things/build/cmd/osx/make HOST_OS=OSX TOOLS_ROOT=/Users/mac/github/AliOS-Things/build -e -f build/Makefile tinyengine_app@developerkit" in "/Users/mac/github/AliOS-Things"
```
**Describe the solution you'd like**
是不是文件缺失了

**Additional context**

xiaowenxia, 2019-01-30:
@chenyulun 目前master分支编译tinyengine确实存在问题，developer分支编译没有问题。

chenyulun, 2019-01-31:
@xiaowenxia 
```aos make tinyengine_app@esp8266```
这个命令在developer里面也报错，mqtt的问题，
```shell
C语言报错：'for' loop initial declarations are only allowed in C99 mode
```
修改了这个后面，生成
Making tinyengine_app@esp8266.elf也有问题；
```
out/tinyengine_app@esp8266/libraries/tiny_engine.a(tiny_engine_main.o):(.literal.tiny_engine_load_addon+0x4): undefined reference to `module_pwm_register'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(tiny_engine_main.o):(.literal.tiny_engine_load_addon+0x8): undefined reference to `module_adc_register'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(tiny_engine_main.o):(.literal.tiny_engine_load_addon+0xc): undefined reference to `module_dac_register'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(tiny_engine_main.o):(.literal.tiny_engine_load_addon+0x10): undefined reference to `module_wdg_register'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(tiny_engine_main.o):(.literal.tiny_engine_load_addon+0x14): undefined reference to `module_rtc_register'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(tiny_engine_main.o): In function `tiny_engine_load_addon':
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/apps/src/tiny_engine_main.c:50: undefined reference to `module_pwm_register'
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/apps/src/tiny_engine_main.c:51: undefined reference to `module_adc_register'
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/apps/src/tiny_engine_main.c:53: undefined reference to `module_dac_register'
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/apps/src/tiny_engine_main.c:54: undefined reference to `module_wdg_register'
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/apps/src/tiny_engine_main.c:55: undefined reference to `module_rtc_register'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(module_gpio.o):(.literal.gpio_module_handle_cb+0x64): undefined reference to `hal_gpio_init'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(module_gpio.o):(.literal.gpio_module_handle_cb+0x68): undefined reference to `hal_gpio_input_get'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(module_gpio.o):(.literal.gpio_module_handle_cb+0x6c): undefined reference to `hal_gpio_output_high'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(module_gpio.o):(.literal.gpio_module_handle_cb+0x70): undefined reference to `hal_gpio_output_low'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(module_gpio.o):(.literal.gpio_module_handle_cb+0x74): undefined reference to `hal_gpio_enable_irq'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(module_gpio.o): In function `gpio_open':
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/gpio/module_gpio.c:51: undefinedreference to `hal_gpio_init'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(module_gpio.o): In function `gpio_read':
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/gpio/module_gpio.c:169: undefined reference to `hal_gpio_input_get'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(module_gpio.o): In function `gpio_write':
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/gpio/module_gpio.c:128: undefined reference to `hal_gpio_output_high'
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/gpio/module_gpio.c:128: undefined reference to `hal_gpio_output_low'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(module_gpio.o): In function `gpio_on':
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/gpio/module_gpio.c:248: undefined reference to `hal_gpio_enable_irq'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(module_gpio.o): In function `gpio_irq':
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/gpio/module_gpio.c:83: undefinedreference to `hal_gpio_input_get'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(sensor_dht11.o): In function `dht11_wait_status':
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/hw/sensor_dht11.c:26: undefined reference to `hal_gpio_input_get'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(sensor_dht11.o): In function `dht11_get_temphum':
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/hw/sensor_dht11.c:117: undefinedreference to `hal_gpio_output_low'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(sensor_dht11.o): In function `dht11_read_byte':
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/hw/sensor_dht11.c:62: undefined reference to `hal_gpio_output_high'
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/hw/sensor_dht11.c:64: undefined reference to `hal_gpio_input_get'
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/hw/sensor_dht11.c:66: undefined reference to `hal_gpio_input_get'
out/tinyengine_app@esp8266/libraries/tiny_engine.a(sensor_dht11.o): In function `dht11_read_bit':
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/hw/sensor_dht11.c:39: undefined reference to `hal_gpio_input_get'
[... cut ...]
out/tinyengine_app@esp8266/libraries/tiny_engine.a(module_i2c.o): In function `i2c_write':
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/i2c/module_i2c.c:129: undefined reference to `hal_i2c_master_send'
/Users/chenyulun390/github/AliOS-Things/framework/tinyengine/bone-framwork/bone-engine/native-modules/i2c/module_i2c.c:154: undefined reference to `hal_i2c_finalize'
collect2: error: ld returned 1 exit status
make[1]: *** [out/tinyengine_app@esp8266/binary/tinyengine_app@esp8266.elf] Error 1
make: *** [main_app] Error 2
```

chenyulun, 2019-01-31:
他说加这个参数怎么加
```
framework/tinyengine/bone-framwork/bone-engine/native-modules/mqtt/module_mqtt.c: In function 'module_handle_cb':
framework/tinyengine/bone-framwork/bone-engine/native-modules/mqtt/module_mqtt.c:181:13: error: 'for' loop initial declarations are only allowed in C99 mode
             for (int i = 0; i < sizeof(k_ipad); i++)
             ^
framework/tinyengine/bone-framwork/bone-engine/native-modules/mqtt/module_mqtt.c:181:13: note: use option -std=c99 or -std=gnu99 to compile your code
```
-std=c99 or -std=gnu99

xiaowenxia, 2019-02-01:
目前tinyengine仅支持developerkit和esp32，不支持esp8266哦，资源受限，可以参考tinyengine的github：https://github.com/aliyun/TinyEngine

closed 2022-03-01 by skylarCai