alibaba/AliOS-Things #852: 编译ldapp@developerkit提示缺少iot_import.h头文件

408881465, 2019-02-06:
**Describe the bug**
> Executing task: aos make ldapp@developerkit <
aos-cube version: 0.2.66
Check if required tools for developerkit exist
Build AOS Now
TOOLCHAIN_PATH=D:/aliosThings/AliOS-Things/build/compiler/gcc-arm-none-eabi/Win32/bin/
app/example/ldapp/mqtt-example.c:9:10: fatal error: iot_import.h: No such file or directory
 #include "**iot_import.h**"
          ^~~~~~~~~~~~~~
compilation terminated.
make.exe[1]: *** [build/aos_target_build.mk:270: out/ldapp@developerkit/Modules/app/example/ldapp/mqtt-example.o] Error 1
make.exe[1]: *** Waiting for unfinished jobs....
make.exe: *** [build/Makefile:274: main_app] Error 2
[AliOS-Things] ERROR: "D:/aliosThings/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "D:/aliosThings/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=D:/aliosThings/AliOS-Things/build -e -f build/Makefile ldapp@developerkit" in "D:\aliosThings\AliOS-Things"

**environment**
1. branch(version): V2.0.0
2. windows 7 VScode:
3. example:

xiaowenxia, 2019-02-12:
developerkit请使用[developer分支](https://github.com/alibaba/AliOS-Things/tree/developer)，2.0.0分支目前developerkit的部分app还是存在问题的。

408881465, 2019-02-12:
谢谢

closed 2019-02-12 by 408881465