alibaba/AliOS-Things #854: Lose Keil libmbedtls.lib for Cortex-M0?

wosayttn, 2019-02-12:
**Is your feature request related to a problem? Please describe.**
請問有 libmbedtls.lib for Cortex-M0 MCU 預先編譯的版本嗎?

**Describe the solution you'd like**
能否提供一下Cortex-M0的。
EX: https://github.com/alibaba/AliOS-Things/tree/master/security/mbedtls/lib/Cortex-M4

**Additional context**
make.exe[1]: *** No rule to make target 'security/mbedtls/lib/Cortex-M0/KEIL/libmbedtls.lib', needed by 'out/mqttapp@numaker-pfm-m2351/binary/mqttapp@numaker-pfm-m2351.axf'.  Stop.

xiaowenxia, 2019-02-12:
@wosayttn  mbedtls是开源的，你可以直接使用官方的[mbedtls源代码](https://tls.mbed.org/)编译一个Cortex-M0 的库使用。:smile:

wosayttn, 2019-03-06:
能否提供一下mbedtls keil 工程檔案 for CM4?

xiaowenxia, 2019-03-07:
> 能否提供一下mbedtls keil 工程檔案 for CM4?

使用命令：
```
aos make app@board IDE=KEIL
```
可以到处keil工程，不过目前有些board还不支持keil工程。

wosayttn, 2019-03-12:
我的需求是 有沒有編譯出 libmbedtls.lib 的 keil 工程檔案? 不是 aos make app@board IDE=keil。
Cortex-M0是沒有預編譯好的libmbedtls.lib，官方建議我們自行編譯加入使用，請問如何編譯它? 需要的版本號?

xiaowenxia, 2019-03-12:
@wosayttn 把 [security/mbedtls/src/](https://github.com/alibaba/AliOS-Things/tree/rel_2.1.0/security/mbedtls/src) 下面的源码放在keil工程中，编译成`library`应该就可以了。

closed 2022-03-01 by skylarCai