alibaba/AliOS-Things #855: 对于aos_test.c中的任务栈溢出问题

sukangjian, 2019-02-12:
[image]
[image]
[image]

sukangjian, 2019-02-12:
程序执行创建任务task1成功后，未进入task1函数中，打印任务栈溢出，程序跑飞；不清楚为什么程序会在这里打印任务栈溢出，将stack_size加大到6K还是打印任务栈溢出

xiaowenxia, 2019-02-12:
能不能描述的再详细一些：
1.  哪个分支？
2.  哪个board？

sukangjian, 2019-02-12:
> 能不能描述的再详细一些：
> 
> 1. 哪个分支？
> 2. 哪个board？

AliOS-Things/test/testcase/certificate_test/aos_test.c
M0核armv6m

xiaowenxia, 2019-02-12:
哪个分支呢？ master还是rel_1.3.3还是其他的分支？

在 board/ 目录里面对应哪个板子型号呢？

sukangjian, 2019-02-12:
> 哪个分支呢？ master还是rel_1.3.3还是其他的分支？
> 
> 在 board/ 目录里面对应哪个板子型号呢？
rel_1.3.0,STM32L073RZ-Nucleo

xiaowenxia, 2019-02-12:
好的。

sukangjian, 2019-02-15:
问题已解决，原因是创建测试任务的栈的值设置的太小，导致测试时任务栈溢出

closed 2019-02-15 by sukangjian