alibaba/AliOS-Things #856: 在vscode中编译linkkitapp@esp32devkitc例程失败

lss6378, 2019-02-13:
**Describe the bug**
在vscode中编译linkkitapp@esp32devkitc例程失败
但是编译linkkitapp@esp8266例程能够通过

**environment**
1. branch(version): developer
2. Desktop OS(windows linux or macOS): Win10
3. example: linkkitapp@esp32devkitc

**linkkitapp@esp32devkitc编译Log如下**

```
> Executing task: aos make linkkitapp@esp32devkitc <

aos-cube version: 0.2.64
Check if required tools for esp32devkitc exist
信息: 用提供的模式无法找到文件。
make.exe: *** No rule to make target 'app/example/linkkitapp/linkkitapp.mk', needed by 'out/linkkitapp@esp32devkitc/config.mk'.  Stop.
[AliOS-Things] ERROR: "F:/code/AliOS/AliOS-Things/build\cmd/win32/make.exe" returned error code 2.
[AliOS-Things] ERROR: Command "F:/code/AliOS/AliOS-Things/build\cmd/win32/make.exe HOST_OS=Win32 TOOLS_ROOT=F:/code/AliOS/AliOS-Things/build -e -f build/Makefile linkkitapp@esp32devkitc" in "F:\code\AliOS\AliOS-Things"
---
The terminal process terminated with exit code: 2

Terminal will be reused by tasks, press any key to close it.
```

**linkkitapp@esp8266编译Log如下**

```
> Executing task: aos make linkkitapp@esp8266 <

aos-cube version: 0.2.64
Check if required tools for esp8266 exist
信息: 用提供的模式无法找到文件。
Build AOS Now
TOOLCHAIN_PATH=F:/code/AliOS/AliOS-Things/build/compiler/gcc-xtensa-lx106/Win32/bin/
信息: 用提供的模式无法找到文件。
Compiling auto_component
Making out/linkkitapp@esp8266/libraries/auto_component.a
Making linkkitapp@esp8266.elf

Making linkkitapp@esp8266.bin
Making linkkitapp@esp8266.hex
no definite address hint,using default mem configuration

                        AOS MEMORY MAP
|=================================================================|
| MODULE                                   | ROM       | RAM      |
|=================================================================|
| newlib_stub                              | 0         | 381      |
| alicrypto                                | 0         | 52474    |
| kv                                       | 0         | 2967     |
| fota                                     | 0         | 2674     |
| socket_stand                             | 0         | 256      |
| alcs                                     | 0         | 17424    |
| vcall                                    | 0         | 2730     |
| kernel_init                              | 0         | 51       |
| yloop                                    | 0         | 1489     |
| rhino                                    | 0         | 17965    |
| framework                                | 0         | 204      |
| libpp                                    | 0         | 62417    |
| digest_algorithm                         | 0         | 2321     |
| libgcc                                   | 0         | 1624     |
| dm                                       | 0         | 26675    |
| fota_download                            | 0         | 22       |
| linkkitapp                               | 0         | 15129    |
| activation                               | 0         | 1369     |
| libmain                                  | 0         | 30209    |
| vfs                                      | 0         | 3363     |
| esp8266                                  | 0         | 9217     |
| fota_mqtt_transport                      | 0         | 2857     |
| board_esp8266                            | 0         | 308      |
| download_http                            | 0         | 1398     |
| ywss4linkkit                             | 0         | 38347    |
| libcirom                                 | 0         | 47163    |
| mqtt                                     | 0         | 14892    |
| cm                                       | 0         | 12893    |
| fota_platform                            | 0         | 128      |
| libnet80211                              | 0         | 45783    |
| xtensa                                   | 0         | 6882     |
| libhal                                   | 0         | 37       |
| link-coap                                | 0         | 11940    |
| mbedtls                                  | 0         | 32704    |
| cjson                                    | 0         | 2521     |
| espos                                    | 0         | 1008     |
| netmgr                                   | 0         | 2080     |
| iotkit                                   | 0         | 31286    |
| log                                      | 0         | 227      |
| MQTTPacket                               | 0         | 2555     |
| hal                                      | 0         | 510      |
| libwpa                                   | 0         | 17894    |
| liblwip                                  | 0         | 45553    |
| libcrypto                                | 0         | 14218    |
| libphy                                   | 0         | 48615    |
| *fill*                                   | 0         | 4430     |
|=================================================================|
| TOTAL (bytes)                            | 0         | 637190   |
|=================================================================|
bin crc: 31a8f8f7
Build complete
Making .gdbinit

Terminal will be reused by tasks, press any key to close it.
```

xiaowenxia, 2019-02-13:
developer分支是能够正常编译`linkkitapp@esp32devkitc`，你`aos make clean`一下再试试能不能编译？

lss6378, 2019-02-13:
> 
> 
> developer分支是能够正常编译`linkkitapp@esp32devkitc`，你`aos make clean`一下再试试能不能编译？

执行 <kbd>aos make clean</kbd> 后，重新编译通过，非常感谢解答

closed 2019-02-13 by lss6378