alibaba/AliOS-Things #857: Lose activation.a for Cortex-M0?

wosayttn, 2019-02-13:
**Is your feature request related to a problem? Please describe.**
請問有 activation.a for Cortex-M0 MCU 預先編譯的版本嗎?

**Describe the solution you'd like**
能否提供一下Cortex-M0的版本?
https://github.com/alibaba/AliOS-Things/tree/master/tools/activation

**Additional context**
out/mqttapp@numaker-pfm-m2351/libraries/libiot_mqtt.a(mqtt_client.o): In function `iotx_mc_report_aos_version':
D:\AliOS-Things/middleware/linkkit/sdk-c/src/protocol/mqtt/./client/mqtt_client.c:2843: undefined reference to `aos_get_version_info'
collect2.exe: error: ld returned 1 exit status
make.exe[1]: *** [build/aos_target_build.mk:338: out/mqttapp@numaker-pfm-m2351/binary/mqttapp@numaker-pfm-m2351.elf] Error 1

xiaowenxia, 2019-02-13:
@wosayttn 你想要哪个工具链的库文件呢？

blackboxer123, 2019-02-13:
activation.a将会在后续版本中删除，上面的编译问题，可以通过将调用 aos_get_version_info 的地方注释掉来解决。不会影响功能。

closed 2022-03-01 by skylarCai