alibaba/AliOS-Things #865: Configure your IntelliSense settings to help find missing headers.

wanghannccandriod, 2019-02-19:
Configure your IntelliSense settings to help find missing headers.?

求助，怎么配置，或是怎么解决，谢谢

xiaowenxia, 2019-02-20:
这个是vscode [c/c++](https://github.com/Microsoft/vscode-cpptools)插件的静态检查哦，一般情况下不需要关注的，对编程没有影响，如果你一定要解决这个提示的话，可以参考这个文档配置一下：[Getting started with IntelliSense configuration](https://github.com/Microsoft/vscode-cpptools/blob/master/Documentation/Getting%20started%20with%20IntelliSense%20configuration.md)

wanghannccandriod, 2019-02-20:
> 这个是vscode [c/c++](https://github.com/Microsoft/vscode-cpptools)插件的静态检查哦，一般情况下不需要关注的，对编程没有影响，如果你一定要解决这个提示的话，可以参考这个文档配置一下：[Getting started with IntelliSense [...]

closed 2019-02-20 by wanghannccandriod