alibaba/AliOS-Things #869: 32位linux下编译一直不顺利

siberianfox, 2019-02-22:
我在使用其他两台64位linux的电脑进行编译和烧录都没有问题，但是换到一台32位linux的笔记本上就一直无法编译。有几个问题：

1.每次执行编译都会重复下载编译工具链，就算文件夹内已经有对应的编译工具链也一样检测不到。
例如：
```
aos-cube version: 0.2.66
Check if required tools for esp8266 exist
toolchain gcc-xtensa-lx106 missing, start download ...
```

2.每次都编译失败，就算执行aos make clean也不行。
例如:
```
Build AOS Now
TOOLCHAIN_PATH=/home/lin/AliOS-Things/build/compiler/gcc-xtensa-lx106/Linux32/bin/
Compiling helloworld
Compiling board_esp8266
Compiling esp8266
/home/lin/AliOS-Things/build/compiler/gcc-xtensa-lx106/Linux32/bin/xtensa-lx106-elf-gcc: 1: /home/lin/AliOS-Things/build/compiler/gcc-xtensa-lx106/Linux32/bin/xtensa-lx106-elf-gcc: Syntax error: Unterminated quoted string
make[1]: *** [out/helloworld@esp8266/Modules/board/esp8266/board.o] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/lin/AliOS-Things/build/compiler/gcc-xtensa-lx106/Linux32/bin/xtensa-lx106-elf-gcc: 1: /home/lin/AliOS-Things/build/compiler/gcc-xtensa-lx106/Linux32/bin/xtensa-lx106-elf-gcc: /home/lin/AliOS-Things/build/compiler/gcc-xtensa-lx106/Linux32/bin/xtensa-lx106-elf-gcc: 1: /home/lin/AliOS-Things/build/compiler/gcc-xtensa-lx106/Linux32/bin/xtensa-lx106-elf-gcc: Syntax error: Unterminated quoted string
Syntax error: Unterminated quoted string
make[1]: *** [out/helloworld@esp8266/Modules/app/example/helloworld/helloworld.o] Error 2
make[1]: *** [out/helloworld@esp8266/Modules/./platform/mcu/esp8266/aos/soc_impl.o] Error 2
/home/lin/AliOS-Things/build/compiler/gcc-xtensa-lx106/Linux32/bin/xtensa-lx106-elf-gcc: 1: /home/lin/AliOS-Things/build/compiler/gcc-xtensa-lx106/Linux32/bin/xtensa-lx106-elf-gcc: Syntax error: Unterminated quoted string
make[1]: *** [out/helloworld@esp8266/Modules/./platform/mcu/esp8266/aos/hook_impl.o] Error 2
make: *** [main_app] Error 2
[AliOS-Things] ERROR: "/home/lin/AliOS-Things/build/cmd/linux32/make" returned error code 2.
[AliOS-Things] ERROR: Command "/home/lin/AliOS-Things/build/cmd/linux32/make HOST_OS=Linux32 TOOLS_ROOT=/home/lin/AliOS-Things/build -e -f build/Makefile helloworld@esp8266" in "/home/lin/AliOS-Things"
---
```

能帮忙看看吗，谢谢!

xiaowenxia, 2019-02-23:
看起来aos-cube下载的`xtensa-lx106-elf-gcc`这个工具链不支持在32位linux上运行。你可以到官网下载一个32位的xtensa工具链放到`build/compiler/gcc-xtensa-lx106/Linux32/`里面。

siberianfox, 2019-02-25:
> 看起来aos-cube下载的`xtensa-lx106-elf-gcc`这个工具链不支持在32位linux上运行。你可以到官网下载一个32位的xtensa工具链放到`build/compiler/gcc-xtensa-lx106/Linux32/`里面。

我尝试了这个方法，但是没起作用，aos还是自己重新下载编译工具，然后又跟上面一样，失败了。另外，其实不止是ESP8266无法编译，基本上所有的架构或者MCU都不行，成功的只有一个helloworld@linuxhost。

xiaowenxia, 2019-02-25:
@siberianfox aos-cube会去判断toolchain的版本号，不匹配会重新下载toolchain。解决办法：
    
打开aos-cube模块的安装目录下的`constant.py`，更改xtensa-lx106 下面的`version`  为 **all** 就行了。
```py
{
    .......
    'xtensa-lx106':{
        'name': 'gcc-xtensa-lx106',
        'path': 'build/compiler/gcc-xtensa-lx106',
        'command': 'xtensa-lx106-elf-gcc',
        'version': 'all',
        'use_global': True,
        'Win32_url': 'https://gitee.com/alios-things/gcc-xtensa-lx106-win32.git',
        'Linux32_url': 'https://gitee.com/alios-things/gcc-xtensa-lx106-linux.git',
        'Linux64_url': 'https://gitee.com/alios-things/gcc-xtensa-lx106-linux.git',
        'OSX_url': 'https://gitee.com/alios-things/gcc-xtensa-lx106-osx.git',
    },
    ......
}
```

谢谢你的反馈。:smile:

closed 2019-03-16 by siberianfox