alibaba/AliOS-Things #871: VS Code IDE 中，执行-exec c 命令， 代码调试区没有切换到相应的断点处

wizephen, 2019-02-23:
1. VS code （Windows 版本：1.21.1）启动调试后，代码调试区可以停止在application_start （跟AliOS Things 代码版本无关）;
2. 在“调试控制台”中， 通过“-exec b function_xxx“”，设置function breakpoint 后，执行“-exec c“ 后，
    “调试控制台”中，可以到GDB 已停止在的对应的断点处，但**代码区**，却无法切换到相应的断点函数 
     中，如下图所示；
[image]
3. “调试控制台”执行“-exec c” 后，在GDB 运行至断点前，浮动调试栏位**没有**禁止部分调试功能，如下图中，绿色框部分在期间应该称不可使用状态；
[image]

xiaowenxia, 2019-02-23:
@wizephen  第二点：

xiaowenxia, 2019-02-23:
@wizephen   这个应该是c/c++插件的debug机制的问题，你通过命令设置bp，应该是反馈不到c/c++插件上的，这样c/c++插件也不会跳转到对应代码区。运行`-exec c`也是一样的道理。操作最好在vscode里面进行，或者在`launch.json` 的 `setupcommands`中设置。

walker2048, 2019-02-24:
如果习惯GDB命令行操作，可以考虑直接用命令行来debug

wizephen, 2019-02-25:
@xiaowenxia 所以这个问题暂时没有办法解决，对么 ？ 那这样的话，我就close 掉这个issue

xiaowenxia, 2019-02-25:
@wizephen 是的，这个问题你可以到c/c++插件的github上提issues问一问能不能解决哈：https://github.com/Microsoft/vscode-cpptools

wizephen, 2019-02-25:
@xiaowenxia OK, 找个时间我去Microsoft github上面complain 一下，嘎嘎!

closed 2019-02-25 by wizephen