alibaba/AliOS-Things #875: alios 的linkkitapp，使用的linkkit的api是V2.3新版的还是旧版的？

scarleast, 2019-02-25:
**Describe the bug**

**environment**
1. branch(version): rel2.0
2. Desktop OS(windows linux or macOS):Mac
3. example: 

想问下，rel2.0 linkkitapp使用的linkkit的api，是V2.3新版的还是旧版的，然后alios stutio在哪里可以查看全局的宏定义。

scarleast, 2019-02-25:
[image]

scarleast, 2019-02-25:
[image]
我看到linkkitapp中，/app/example/linkkitapp/linkkit_example_solo.c的文件有这样的预编译选项，但是关于全局宏变量DEPRECATED_LINKKIT并没有找到它在哪里定义了，但是编译的过程中，确实认为定义过这个变量的，所以很好奇，alios stutio在哪里可以查看全部的宏定义。

xiaowenxia, 2019-02-26:
@scarleast `DEPRECATED_LINKKIT`这个宏 是在 `linkkitapp.mk`中设置的，看这里：https://github.com/alibaba/AliOS-Things/blob/rel_2.0.0/app/example/linkkitapp/linkkitapp.mk#L31

scarleast, 2019-02-26:
ifeq ($(case),sched) 这里的case变量我看在mk文件中并没有定义，sched这个值又代表的什么？不好意思哈，这块确实看的挺迷糊的。

xiaowenxia, 2019-02-26:
这个是输入编译指令的时候指定的：

```
aos make linkkitapp@developerkit case=sched
```

scarleast, 2019-02-26:
[image]

我使用alios studio 编译的，默认编译aos指令好像没有case=sched。。。

xiaowenxia, 2019-02-26:
@scarleast 是的，默认没有指定，可以编辑`.vscode/tasks.json` 下面的`alios-studio: Make`，把参数添加进去：
```json
  {
      "label": "alios-studio: Make",
      "type": "shell",
      "command": "aos",
      "args": [
        "make",
        "helloworld@developerkit",
        "case=sched"
      ],
      "presentation": {
        "focus": true
      }
    }
```

scarleast, 2019-02-26:
非常感谢，明白了

closed 2019-02-26 by scarleast